meshtastic/firmware #2541: [Bug]: TTGO LILYGO LORA32 T3_V1.6.1 with FW 2.1.11 ends up in endless boot cycle

Elr0n, 2023-06-05:
### Category

Other

### Hardware

T-Lora v1

### Firmware Version

2.1.11.5ec624d beta

### Description

Reproduceable (with 2 TTGO LILYGO LORA32 T3_V1.6.1 with FW 2.1.11) after 2-3 days of continuous operation (USB Power Supply), the device ends up "frozen" / in an endless reboot cycle, just showing the Meshtastic Start/Splash Screen. See related log output attached.

### Relevant log output

```Shell
ELF file SHA256: 354ecff2c27c8d56

E (10135) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (817) esp_core_dump_flash: No core dum ⸮⸮ѥѥ⸮⸮⸮found!
E (817) esp_core_dump_flash: No core dump partition found!
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
⸮⸮ @INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (487424/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (90 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (7174 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (71 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (592 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (600 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3162 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2221 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (842 Bytes)
DEBUG | ??:??:?? 0  /static/index-0e816946.js.gz (399863 Bytes)
DEBUG | ??:??:?? 0  /static/index-9602d760.css.gz (17494 Bytes)
DEBUG | ??:??:?? 0  /static/index-fc2e9253.js.gz (347 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (196 Bytes)
[   405][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
DEBUG | ??:??:?? 0 Using analog input 35 for battery level
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   421][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
INFO  | ??:??:?? 0 ssd1306 display found
INFO  | ??:??:?? 0 ssd1306 display found
DEBUG | ??:??:?? 0 0x7 subtype probed in 2 tries 
INFO  | ??:??:?? 0 1 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 Meshtastic hwvendor=3, swver=2.1.11.5ec624d
DEBUG | ??:??:?? 0 Setting random seed 2307366781
DEBUG | ??:??:?? 0 Total heap: 192936
DEBUG | ??:??:?? 0 Free heap: 159060
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 110, FreeEntries 520, AllEntries 630, NameSpaces 3
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 890
DEBUG | ??:??:?? 0 OTA firmware version 0.2.1.2fbe7fc
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 20
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded saved config version 20
INFO  | ??:??:?? 0 Loading /prefs/module.proto
INFO  | ??:??:?? 0 Loaded saved moduleConfig version 20
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 20
[   643][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 0 Number of Device Reboots: 890
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 3, using EU_868
DEBUG | ??:??:?? 0 region=3, NODENUM=0x25b1e384, dbsize=80
INFO  | ??:??:?? 0 Saving /prefs/db.proto
DEBUG | ??:??:?? 0 Read RTC time as 1685965163
DEBUG | ??:??:?? 0 Using MSL altitude model
DEBUG | ??:??:?? 0 WANT GPS=1
DEBUG | ??:??:?? 1 Module Info : 
DEBUG | ??:??:?? 1 Soft version: .00 (59842)
DEBUG | ??:??:?? 1 Hard version: 0070000
DEBUG | ??:??:?? 1 Extensions:2
DEBUG | ??:??:?? 1   ROTVER 14.00
DEBUG | ??:??:?? 1   PS;SBAS;GLO;QZSS
INFO  | ??:??:?? 1 UBlox GNSS init succeeded, using UBlox GNSS Module
DEBUG | ??:??:?? 1 GxGSA NOT available
INFO  | ??:??:?? 1 External Notification Module Disabled
INFO  | ??:??:?? 1 Turning on screen
DEBUG | ??:??:?? 2 Module wants a UI Frame
DEBUG | ??:??:?? 2 Module wants a UI Frame
WARN  | ??:??:?? 2 RF95Interface(cs=18, irq=26, rst=23, busy=33)
INFO  | ??:??:?? 2 Starting meshradio init...
DEBUG | ??:??:?? 2 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=0, time -1 ms
DEBUG | ??:??:?? 2 (bw=0, sf=0, cr=4/0) packet symLen=2147483647 ms, payloadSize=253, time -1 ms
INFO  | ??:??:?? 2 Radio freq=869.400, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 2 Set radio: region=EU_868, name=Custom, config=0, ch=-1397824096, power=27
INFO  | ??:??:?? 2 Radio myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 mhz)
INFO  | ??:??:?? 2 Radio myRegion->numChannels: -1 x 0.000kHz
INFO  | ??:??:?? 2 Radio channel_num: -1397824095
INFO  | ??:??:?? 2 Radio frequency: 869.400024
INFO  | ??:??:?? 2 Slot time: 42 msec
INFO  | ??:??:?? 2 Set radio: final power level=20
DEBUG | ??:??:?? 2 Current limit set to 100.000000
DEBUG | ??:??:?? 2 Current limit set result 0
INFO  | ??:??:?? 2 RF95 init result -8
INFO  | ??:??:?? 2 Frequency set to 869.400024
INFO  | ??:??:?? 2 Bandwidth set to 0.000000
INFO  | ??:??:?? 2 Power output set to 20
WARN  | ??:??:?? 2 Failed to find RF95 radio
INFO  | ??:??:?? 2 Not using WIFI
ERROR | ??:??:?? 2 NOTE! Recordinɥѥ⸮⸮⸮⸮error 3 at src/main.cpp:628
INFO  | ??:??:?? 2 PowerFSM init, USB power=1
INFO  | ??:??:?? 2 Enter state: BOOT
[  2118][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4021a9e5  PS      : 0x00060630  A0      : 0x800e21b4  A1      : 0x3ffd8b40  
A2      : 0x3ffd9690  A3      : 0x00000140  A4      : 0x00000001  A5      : 0x3f42587d  
A6      : 0x3f4255f7  A7      : 0x00000846  A8      : 0x801409b9  A9      : 0x3ffd8b30  
A10     : 0x00000000  A11     : 0x3ffd8b5f  A12     : 0x3ffcf71c  A13     : 0x00060623  
A14     : 0x00060620  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40093aed  LEND    : 0x40093afd  LCOUNT  : 0xffffffff  

Backtrace: 0x4021a9e2:0x3ffd8b40 0x400e21b1:0x3ffd8b70 0x400dfe1c:0x3ffd8ba0 0x40132dc9:0x3ffd8bc0
```

caveman99, 2023-06-05:
Looks like your radio chip is crashing... `INFO  | ??:??:?? 2 RF95 init result -8`

The actual crash is here: 

```
    0x4021a9e2: Router::getQueueStatus() at /home/runner/work/firmware/firmware/src/mesh/Router.cpp:157
    0x400e21b1: MeshService::loop() at /home/runner/work/firmware/firmware/src/mesh/MeshService.cpp:94
    0x400dfe1c: loop() at /home/runner/work/firmware/firmware/src/main.cpp:700
    0x40132dc9: loopTask(void*) at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```
Which is understandable when you don't have a radio interface.

Maybe try the latest alpha? .11 is kinda weeks old by now :-)

Elr0n, 2023-06-05:
Thx for the quick reply and the hint with the radio chip!

I am already busy flashing all my devices to the latest version and will monitor the behavior afterwards. Any assumptions what could trigger the failing radio chip init (EM noise, VDD issues, ...)?

Thx a lot for all your great work with this project!

closed 2023-12-16 by garthvh

milkpirate, 2024-01-22:
Hey @caveman99 / @garthvh 

I am having the same device (or at least the sticker on it say so): LILYGO LORA32 T3_V1.6.1, but I dont really know which device to select for flashing:

[image]

Can you help me out with that?

The vendor states this pin out although the photo say v1.6.1 but the caption says v2.1.6 🤷🏻:

[image]