meshtastic/firmware #2543: [Bug]: Static IP configuration doesn't work

micheljung, 2023-06-05:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.1.11

### Description

Connect a RAK with an ethernet module and DHCP enabled, it will successfully get an IP address (e. g. 192.168.1.223) and can be pinged.

Switch to the static IP configuration, set the exact same IP address (192.168.1.223) and appropriate subnet mask (255.255.255.0) and you can no longer ping the device.

garthvh, 2023-06-05:
Are you converting the ip address to an integer?

micheljung, 2023-06-05:
No, I'm setting the IP address via Android App and it enforces x.x.x.x

In the debug panel of the app, it correctly shows the IP configuration in decimal values.

PR #2544 (2023-06-07): Fix static ethernet config
With static ethernet config, `status` stayed `0` which let the function return without setting `ethEvent`. Therefore, `reconnectETH` was never called and network services were never started.

Fixes #2543
commit 194833d7 (2023-06-07): Fix static ethernet config (#2544)

With static ethernet config, `status` stayed `0` which let the function return
without setting `ethEvent`. Therefore, `reconnectETH` was never called and network services were never started.

Also, the RAK4631 uses little endian, which is why the IP addresses need to be
converted before setting them.

Fixes #2543
closed 2023-06-07 by thebentern