meshtastic/firmware #2556: [Feature Request]: Option completely block rebroadcasting messages above a certain RSSI

EternityForest, 2023-06-10:
### Platform

NRF52, ESP32

### Description

Kind of a minor thing, but If just a few people are using handheld devices and moving around a lot, it's possible that everyone could be close together, and there might be no need to repeat, but devices would repeat anyway if there was no distant node to do so.

Even with the dynamic contention window, depending on the specific time window algorithm it's still possible because of randomness that a non-optimal node might send something.

Nodes that don't have some kind of special antenna type or position could sense the RSSI and refrain from retransmitting some very strong signals, if they detect that they are likely in the same room or within 10 feet or so.

GUVWAF, 2023-06-10:
The first rebroadcast also counts as acknowledgment. I think it's not a good user experience if this doesn't work when testing in the same room. For real-life use, I think it's a minor issue and it's hard to define a good RSSI threshold for this; you would probably also want to incorporate SNR. I think there are better ways to reduce traffic, such as lowering the default Device Telemetery or Position updates. 

It's also best to set a node you describe as having "some kind of special antenna type or position" as a [Router or Repeater](https://meshtastic.org/docs/settings/config/device#role), which get priority in rebroadcasting. That would solve having a non-optimal node rebroadcast.

EternityForest, 2023-06-12:
I definitely agree it's a pretty minor thing and the threshold would be hard to define, it would probably have to be set per-board, and possibly made dynamic with the repeater's current TX power setting.

It might be more relevant in combination with my other #2557, if tunneling over BLE ever became a thing, there would be more use cases for shorter links and closer together nodes with tighter power requirements,  if you were doing tinycoinsizedsensor>(3 feet of BLE)>router>(LoRa) type stuff.