meshtastic/firmware #2557: [Feature Request]: Directly mesh via BLE

EternityForest, 2023-06-10:
### Platform

NRF52, ESP32

### Description

It seems like the protocol itself would run just as well via Bluetooth as it would via LoRa, and modern BLE range can be really good.

Meshing directly via Bluetooth would solve several big issues.  For one thing it would let a group share a single Meshtastic node, sending and receiving messages from their phones, without anyone needing to have settings access to the node.

In an emergency situation, I imagine that could be pretty valuable.   Campgrounds outside of cell range could have permanently installed solar Meshtastic nodes that any guest could use with no setup effort needed.

It would also allow for very low cost/low power/tiny sensors along the lines of AirTags.   A single LoRa device could cover a whole area of tiny nodes that send their messages once a day.

Also, even though it might not be part of the core mission, it would let people try out, and make real use of, all the same features without buying new hardware.

Bluetooth also has virtually unlimited bandwidth compared to LoRa, and there is extended advertising allowing for additional features.  

Bluetooth packets could have "Virtual modem settings" and be repeated onto LoRa only if the modem settings matched the node's setup, allowing you to keep packets off LoRa if they only need to be short range, so short and long range applications could share the same code.

charminULTRA, 2023-06-25:
Wouldn't BLE range be significantly shorter? Like 1% the range of LoRa, if that?

EternityForest, 2023-06-25:
BLE's theoretical range is like 500 feet, and even the practical range is
fairly decent.  Nowhere near LoRa, but enough for a lot of use cases,
especially with a BLE to LoRa repeater gateway.

mattdrum, 2024-02-13:
I could see this functionality being useful for crossband repeaters. Additionally, implementing this for two radios on the same network, one being on a directional antenna for a long distance uplink, and a second one being used for omnidirectional coverage.