meshtastic/firmware #2563: [Feature Request]: More Flexible Location Data Configuration

jp-bennett, 2023-06-16:
### Platform

NRF52, ESP32

### Description

Location data is currently sent over the Primary channel, if enabled at all. It's rather difficult to change things around, to send location data to a different channel. When using a Meshtastic radio for multiple purposes, like family tracking and also disaster recovery, a user might want a simple way to change on what channels location data is sent. For normal operation, only the private, primary channel would get location data. But right after a disaster, the location could be turned on for everyone.

I suggest a pair of options for each channel, to send and receive location data. Easily configured by end users, even non-technical users. This would allow meshtastic users to quickly adapt to an emergency, and still manage their privacy during normal times.

jp-bennett, 2023-06-17:
Turning on sending of location data on non-primary channels should trigger a "big annoying warning" that it can cause massive slow-downs to the mesh.

RightHandMan1, 2023-06-21:
I'm kinda with you on this one. It would also solve an MQTT issue where Primary ch dont match over many nodes, but a matching secondary(s) would. Especially when SHTF , no internet is available, and a local laptop w a wifi router and MQTT locally hosted,, is providing comm to a neighborhood or SAR event..  But, there would probably need a number after the peer to indicate which 'channel' the node is currently seen on,etc.

jp-bennett, 2023-06-29:
I've been thinking more about this, and I'm wondering if there's a compelling reason to actually have a "Primary" channel. It seems like a meaning distinction, since channel 0 is always primary.

garthvh, 2023-06-29:
The channel index is an app only construct, it is not sent over the mesh.

garthvh, 2024-03-10:
Got all sorts of new location flexibility

closed 2024-03-10 by garthvh