meshtastic/firmware #2566: [Feature Request]: Indication of app->node connectivity over mesh.

prokrypt, 2023-06-19:
### Platform

NRF52, ESP32

### Description

We can already tell when we have last heard from a node, but it would be great if we could identify over the mesh whether or not a node has an app/client connected to it, so we can see if "anyone's home".

garthvh, 2023-06-19:
Devices are connecting and disconnecting all the time when they go out of range, wifi drops etc, this would generate a ton of mesh traffiic

prokrypt, 2023-06-19:
> Devices are connecting and disconnecting all the time when they go out of range, wifi drops etc, this would generate a ton of mesh traffiic

I was thinking of piggybacking it onto regular telemetry/position/nodeinfo updates rather than having it be event-based, and when there are direct/global messages received we can assume there's a person behind it...

thebentern, 2023-06-21:
This is already account for in the firmware. It can be done manually with a connection status admin message, but it's a manual process and too niche of a use-case for clients to implement. So I wouldn't count on seeing it pop up in Android or iOS anytime soon.

closed 2023-06-21 by thebentern