meshtastic/firmware #2575: [Bug]: MQTT JSON From field wrong value

erstec, 2023-06-26:
### Category

Other

### Hardware

T-Beam, T-Lora v2 1.6

### Firmware Version

2.1.16.a2c5b92

### Description

Packets originating from nodes updated to 2.1.16.a2c5b92 appears on MQTT with `from` field equal 4. Regardless of they real id.
Same time over LoRa they are sane, `--nodes` and Android/iOS apps show them correctly.
And, most probably, non-json version of packets published to MQTT Broker `/msh/2/c` topic are also don't have such problem, as they retransmitted to local mesh correctly.

MQTT JSON topic `/msh/2/json` looks like like:
```{'channel': 0, 'from': 4, 'id': 1060460488, 'payload': {'barometric_pressure': 0, 'current': 0, 'gas_resistance': 0, 'relative_humidity': 0, 'temperature': 27.875, 'voltage': 0.871999979019165}, 'sender': '!d1c7fc8c', 'timestamp': 1687797886, 'to': -1, 'type': 'telemetry'}```

Revert to 2.1.15 solves issue.

GUVWAF, 2023-06-26:
The output from `--nodes` shows the NodeId, which is (slightly) different from NodeNum. Probably if you run `--info` it would display 4 there as well. This stems from another bug that's hopefully fixed by #2576.

erstec, 2023-06-26:
> The output from `--nodes` shows the NodeId, which is (slightly) different from NodeNum. Probably if you run `--info` it would display 4 there as well. This stems from another bug that's hopefully fixed by #2576.

Tried PR build artifacts, working properly, from field contais correct value over MQTT JSON and `msh/2/c` topic, except, somehow, fixed position was lost after update, setting `--selat --setlon --setalt` restored it.

GUVWAF, 2023-06-26:
Thanks for testing.
That setting is actually stored in the NodeDB and probably for the wrong NodeNum then, so it makes sense it was lost.

closed 2023-06-27 by thebentern