meshtastic/firmware #2578: [Bug]: Firmware 2.1.16 does not compile from source code

Burnwald, 2023-06-27:
### Category

Hardware Compatibility

### Hardware

T-Beam S3

### Firmware Version

2.1.16

### Description

I failed to build firmware from default unmodified source code. Also checked with 2.1.15
[image]

caveman99, 2023-06-27:
Please clean up your platformio environment. Clear cache and package directories and let the plugin download the tool chains again. If this was a real problem, CI would also not build.

Burnwald, 2023-06-27:
Thank you for quick reply, that was a new clean install, new project and folder in the studio. Error I received appears after I open folder and wait until all chains are updated. When I try to build I receive more errors:
[image]
Sorry, doesn't have a lot of experience with Visual Studio.

caveman99, 2023-06-27:
I just pushed a fix for the end of file message.
I just successfully build that t<arget with vscode here locally on windows, from Master branch. Please do yourself a favor and start over, following this step by step guide here: https://meshtastic.org/docs/development/firmware/build

closed 2023-06-28 by thebentern