meshtastic/firmware #2581: [Bug]: Remote command is applied localy

kri164, 2023-06-29:
### Category

Other

### Hardware

Other

### Firmware Version

2.1.15.cd78723 (and several randomly selected previous versions)

### Description

When try to set fixed location on remote node, it is applied to local node.

Admin channel is working, setting --set-owner on remote node is working.
When try to set fixed position on remote node, this setting is appllied to local node.

This bug is observed on HW: TT-Beam, TT-Lora-T3S3, RAK4631, Heltec WLS.

### Relevant log output

```Shell
Node address and position is redacted.

(venv) linux:~/MESH # meshtastic --dest '!12345678' --set-owner "Remote"
Connected to radio
Setting device owner to Remote
Waiting for an acknowledgment from remote node (this could take a while)
Received an ACK.
(venv) linux:~/MESH #

(venv) linux:~/MESH # meshtastic --dest '!12345678' --setlat 44.44444 --setlon 22.22222 --setalt 333
Connected to radio
Fixing altitude at 333 meters
Fixing latitude at 44.44444 degrees
Fixing longitude at 22.22222 degrees
Setting device position
(venv) linux:~/MESH #
```

thebentern, 2023-06-29:
Hi. This is an issue with the python CLI, not the firmware. Fixed position cannot be set over the mesh. 
I recommedn that these configurations be setup on a remote node before deployment. Remote administration is intended for small tweaks and reboots, not really provisioning a node largely from scratch.

kri164, 2023-06-29:
This is unpleasant, as the fixed position of node is stored in NodeDB. When NodeDB reset is trigered by mistake, there is no way to recovery configuration by remote.

garthvh, 2023-06-29:
Only the fixed position setting can be sent remotely, the coordinate value is the last position packet it has for device and that is indeed in the nodedb.

closed 2023-06-29 by garthvh