meshtastic/firmware #2589: [Bug]: GPS module not detected, might not work at all in firmware 2.1.16+

tropho23, 2023-07-07:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.1.16

### Description

When ESP32 devices are flashed with firmware v2.1.16 or v2.1.17 (latest version as of July 6, 2023), the GPS module is not detected by the firmware, despite correct electrical connection and TX/RX GPIO configuration.

`meshtastic --info` does not report `hasGPS:true` as it should when GPS module is present, and 'No GPS module' is displayed on the device screen.  This bug is confirmed and can be reproduced on Heltec v3 and Heltec Wireless Stick Lite v3 (WSLv3) devices.

When firmware v2.1.15 is flashed, the firmware correctly detects and reports `hasGPS:true`.  'No sats' or the number of detected satellites are displayed on the device screen.

I have not tested this with a T-Beam v1.1, but another use on Discourse has reported a similar issue: https://meshtastic.discourse.group/t/no-gps-module-in-t-beam-v1-1-w-2-1-17-7-alpha/7842

geeksville, 2023-07-07:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/no-gps-module-in-t-beam-v1-1-w-2-1-17-7-alpha/7842/3

code8buster, 2023-07-07:
We need to implement MON_VER message and parse the reply to determine whether or not the Ublox module actually supports the GPS config messages @D4rk4 has implemented.
from https://meshtastic.discourse.group/t/no-gps-module-in-t-beam-v1-1-w-2-1-17-7-alpha/7842/4 :
```
DEBUG | ??:??:?? 1 WANT GPS=1
WARN  | ??:??:?? 2 No response for class 06 message 3E
WARN  | ??:??:?? 2 Unable to reconfigure GNSS, keep factory defaults
WARN  | ??:??:?? 4 No response for class 06 message 39
WARN  | ??:??:?? 4 Unable to enable interference resistance.
WARN  | ??:??:?? 5 No response for class 06 message 23
WARN  | ??:??:?? 5 Unable to configure extra settings.
WARN  | ??:??:?? 7 No response for class 06 message 01
WARN  | ??:??:?? 7 Unable to disable NMEA GGL.
DEBUG | ??:??:?? 7 GxGSA NOT available
WARN  | ??:??:?? 7 GPS FactoryReset requested
```

thebentern, 2023-07-11:
Adding some clarification to this one...  This statement is irrelavent now as we discussed in Discord. MyNodeInfo.has_gps was deprecated and this does not have any bearing on other behavior
 > meshtastic --info does not report hasGPS:true as it should when GPS module is present, and 'No GPS module' is displayed on the device screen. This bug is confirmed and can be reproduced on Heltec v3 and Heltec Wireless Stick Lite v3 (WSLv3) devices.

D4rk4, 2023-07-11:
@tropho23, the issue you've referenced in the [meshtastic.discourse.group thread](https://meshtastic.discourse.group/t/no-gps-module-in-t-beam-v1-1-w-2-1-17-7-alpha/7842) seems to be a different one than what you're currently experiencing.

Could you please test your module with [this firmware](https://github.com/meshtastic/artifacts/blob/device/pr/pr2606-firmware-2.1.19.818af13.zip?raw=true) and see if the issue persists?

If the problem continues, it would be helpful if you could provide the Meshtastic settings related to your position configuration. This would allow us to better understand your setup and further diagnose the issue.

closed 2023-08-14 by garthvh