meshtastic/firmware #2597: [Bug]: Separate U-Blox Neo-6M (v14) GPS init from M8 (v15) init and recover bitrate

Max-Plastix, 2023-07-08:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.1.18

### Description

In some cases, the Neo-6M GPS found on LILY TTGO T-Beam v1.1 units with SMA does not initialize correctly.

Many of the U-Blox commands sent to it are rejected because they match a later version of the protocol (v15) instead of the v14 that is supported.

If the power-on bitrate is not 9600, the module might never sync correctly or have the correct sentences configured.

I would propose making these changes:

1. Speed u-Blox interface from 9600 to 115200.  Messages arrive faster, with less ongoing data stream burden.
2. Add a baudrate search (try them all) if the module does not connect at 115200 on powerup
3. Detect firmware & protocol version to differentiate M6 from 8M
4. Add protocol-specific initialization to configure messages and save settings
5. (maybe) Add a more thorough "factory reset" fallback to zero out settings and reconfigure.

### Relevant log output

```Shell
DEBUG | ??:??:?? 1 WANT GPS=1
WARN  | ??:??:?? 3 No response for class 06 message 3E
WARN  | ??:??:?? 3 Unable to reconfigure GNSS, keep factory defaults
WARN  | ??:??:?? 4 No response for class 06 message 39
WARN  | ??:??:?? 4 Unable to enable interference resistance.
WARN  | ??:??:?? 6 No response for class 06 message 23
WARN  | ??:??:?? 6 Unable to configure extra settings.
WARN  | ??:??:?? 7 No response for class 06 message 01
WARN  | ??:??:?? 7 Unable to disable NMEA GGL.
DEBUG | ??:??:?? 7 GxGSA NOT available
```

Max-Plastix, 2023-07-08:
I'd welcome any discussion around these proposed changes, as I'm new to Meshtastic but not new to the T-Beam or GPS configuration.   

I'm happy to make a test & PR if this sounds like a move in the right direction.

RandMental, 2023-07-12:
Agree this needs to be done

jurriaan, 2023-08-19:
Agree! I had some issues with the Neo-9M (Beitian BK-357) which uses a 38400 baud rate. With no way of configuring the baud rate in meshtastic I had to use the U-center software to set the module to 9600 baud.

jp-bennett, 2023-08-28:
Working on better GPS detection, including a baudrate scan, here: https://github.com/meshtastic/firmware/pull/2752

Would love to get more eyeballs/ testers on this.

kokroo, 2023-08-31:
> Working on better GPS detection, including a baudrate scan, here: #2752
> 
> Would love to get more eyeballs/ testers on this.

How exactly can we help? I have a unit with a 6M and another one with M8N.

jp-bennett, 2023-08-31:
> How exactly can we help?

Particularly if you have a device where the GPS is set at a different speed, try an image from https://github.com/meshtastic/firmware/actions/runs/6029724453#artifacts

davcx, 2023-08-31:
> Particularly if you have a device where the GPS is set at a different speed, try an image from https://github.com/meshtastic/firmware/actions/runs/6029724453#artifacts

I have 2 Tbeam 1.1 with neo6m stucked.
I uploaded firmware-tbeam-2.2.4.3d2c419 and after almost 20 reboot gps emerged with first sentence [GPS] NMEA GPS time 2023-08-31 ...

Now downgraded to 2.2.3 both Tbeams with 6m gps still working (finger crossed).

[esp-web-tools-logs (3).txt](https://github.com/meshtastic/firmware/files/12489906/esp-web-tools-logs.3.txt)

jp-bennett, 2023-08-31:
> after almost 20 reboot gps emerged with first sentence

Super interesting results, thanks!

jp-bennett, 2023-09-03:
@Max-Plastix I think we're partway there with #2752, which just merged. Feel free to test. I don't have a T-beam with the 6m to try it on. https://github.com/meshtastic/firmware/actions/runs/6066814336#artifacts

davcx, 2023-09-05:
@jp-bennett by my POV the issue has been solved. Thx

closed 2023-09-05 by jp-bennett