meshtastic/firmware #2598: [Feature Request]: T-Beam v1.1 change the charge current

PhilipLykov, 2023-07-09:
### Platform

ESP32

### Description

The T-Beam v1.1 use AXP192 charging chip which has an option to control charge current by I2C. For those who use high capacity batteries such as 21500 or even 2 such batteries (10Ah in total) it could be useful. 

The maximum change current is 1320mA. The default is 780mA but actually on T-Beams I have it is about 500mA.

To realize it we need to change the value of register 33H, first 4 bits (0-3) with the following options:
0000:100mA； 0001:190mA；  0010:280mA；  0011:360mA；  0100:450mA； 0101:550mA；  0110:630mA；  0111:700mA； 1000:780mA； 1001:880mA；  1010:960mA；  1011:1000mA；  1100:1080mA；1101:1160mA； 1110:1240mA； 1111:1320mA.

The full list of option could be find on official datasheet on pages 37-38.
[image]

Max-Plastix, 2023-07-10:
Do you know if the USB Port on the T-Beam negotiates for higher than default power?  I checked the Windows driver, and it appears no.. it's only approved for 500mA (default).  Sure, most USB Hubs won't kick you off for drawing a bit more, but I think if you get over 1A you might at least find laptops that reject the device for overcurrent?  I don't actually know.. just figure there's more to it than cranking up the AXP charge current. to 1800mA maximum.

The Default AXP192 setting is 780mA, if I remember correctly.  I forget why, but I had to dial it down to 550mA to work on some hosts.

`    
axp.setChargeControlCur(AXP1XX_CHARGE_CUR_550MA);  // Default 0x1000 = 780mA, more than we can get from USB
`

PhilipLykov, 2023-07-10:
I never charge it from laptop, always from wall charger which supply up to 3A on 5V.

PhilipLykov, 2023-07-21:
I've changed the power.cpp:
PMU->setChargerConstantCurr(XPOWERS_AXP192_CHG_CUR_450MA);
to 
PMU->setChargerConstantCurr(XPOWERS_AXP192_CHG_CUR_1320MA);

Actualy starting from value of 780 it doesn't have any real effect (780 is an upper limit). Checked with multimeter there is 730-750mA churging current to battery.

garthvh, 2023-07-21:
That breaks usb charging right?

Max-Plastix, 2023-07-21:
This change is really ill advised.  The T-Beam has no USB-PD  or D+/D- negotiation, so making the wild presumption that any VBUS can source 1340mA will exclude some number of USB-A chargers.

450mA is specifically selected because no negotiation is needed, and every USB source can provide it without fanfare.  At 1320mA, many (desktop, laptop) USB-A ports will correctly flag over-current and disconnect.

thebentern, 2023-07-22:
> This change is really ill advised. The T-Beam has no USB-PD or D+/D- negotiation, so making the wild presumption that any VBUS can source 1340mA will exclude some number of USB-A chargers.

> 450mA is specifically selected because no negotiation is needed, and every USB source can provide it without fanfare. At 1320mA, many (desktop, laptop) USB-A ports will correctly flag over-current and disconnect.

This is why we made the concious decision early provide this safer simplification in the first place. If an AXP PMU supports power delivery in the future, we can look at adding support for configuring that in the future. Closing this issue.

closed 2023-07-22 by thebentern

adrianTNT, 2025-06-10:
I need some help and people here seem to know what I need ...
I am trying to power my T-Beam Supreme from a solar charger, something like in image, with T-beam Supreme as consumer on the right side. **I will probably use another charger** because people say that one is not apropiate for solar, but I still have some questions:

- If a similar charger outputs ~5V, can I connect it's output to the T-Beam supreme `VBUS pin` instead of USB-C female?
- Will there be a problem if at some point I also plug an USB-C cable in the T-Beam Supreme while it is already powered by that `VBUS pin`? T-Beam will not sent power from USB-C back to that charger and battery, will it?
-  Wiring a battery or charger  to the `VBAT pin` of the T-Beam is the exact same thing as wiring to the battery terminals? Someone said that wiring a battery to `VBAT` pins directly might bypass certain charging safety functions on the T-Beam.

[image]

[image]