meshtastic/firmware #2602: [Bug]: Device telemetry encodig (INA219 - current and volage sensisng) is wrong. This happens on rels > 2.1.13

vinloren, 2023-07-09:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.1.18

### Description

on Tlora2 -1-1-6 device  the encoding of the INA219 telemetry data is wrong as you can see in the attached screen shot: the voltage value is 22.94V and the current is ‘infinitymA’. This happens also on rel 2.1.15 and 2.1.17, 2.1.13 is working well instead.
[image]

pdxlocations, 2023-07-09:
I can duplicate this bug on tlora V2-1 1.6
[image]

pdxlocations, 2023-07-09:
@thebentern  could be something in #2536 ?

thebentern, 2023-07-09:
> @thebentern  could be something in #2536 ?

That's what I'm thinking

vinloren, 2023-07-10:
It seems there is a relationship with #2536

PR #2613 (2023-07-16): Fix INA sensor bug with no longer initializing in Environment Telemetry
Closes #2602
closed 2023-07-16 by thebentern