meshtastic/firmware #2623: [Bug]: Whenever a text message is sent the device reboots and the current BT connection is lost

vinloren, 2023-07-18:
### Category

BLE, Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.1.19.ab325036

### Description

The text message induces a reboot while actually the messages was not sent yet, the BT connection is lost and if I establish the BT connection again it stays on only if the queued message is sent nicely otherwise a new boot sequence starts again and the BT connection is lost. This sequence can go on even for 5 or 6 times until the text message sending succeede.

mverch67, 2023-07-18:
Is this a local build? I can't find this firmware version in the artifacts repository.
If so pls do a git pull and rebuild, then this error should not happen anymore.

code8buster, 2023-07-18:
Try a recent pull request artifact? https://github.com/meshtastic/artifacts/blob/device/pr/pr2621-firmware-2.1.19.985690b.zip?raw=true
This should have mverch's NimBLE fix.

vinloren, 2023-07-18:
I just pulled the this morning updated 2.1.19.8927cffd and the problem is solved. Closing issue.

closed 2023-07-18 by vinloren