meshtastic/firmware #2641: [Feature Request]: Serial mode txtmsg supporting multiple chats

andi448, 2023-07-24:
### Platform

NRF52, ESP32

### Description

Would be great to have the txtmsg mode supporting to write to multiple chats and not limited to one channel only.

GUVWAF, 2023-09-23:
The `TEXTMSG` mode is supposed to be a very simple mode. If we add this, it becomes harder to use, while you still don’t have full control. Others have requested e.g. the ability to send a direct message to a specific destination. All this should be possible using the `PROTO` mode. I opened a Pull Request to the Arduino client library (https://github.com/meshtastic/Meshtastic-arduino/pull/7) with an example how to do this. Could you maybe try if you can make it work using this?

GUVWAF, 2023-09-26:
Please use the `PROTO` mode, e.g. with the updated [Arduino client library](https://github.com/meshtastic/Meshtastic-arduino) to accomplish this instead.

closed 2023-09-26 by GUVWAF