meshtastic/firmware #2646: [Feature Request]: Improve modem preset naming convention

charminULTRA, 2023-07-25:
### Platform

NRF52, ESP32

### Description

This is an old concept I had last year that I believe could use a revive. We currently use things like "Long-Fast" and "Medium-Slow" which are naming conventions that appear on their face to reflect the 1) distance and 2) latency of the presets independently, but in reality do not have a direct relationship to these variables. Rather, the continuum of presets affects both distance and latency at the same time. I believe this relationship should be made more clear via a change in the naming convention.

Good news is that I believe last year we did update the docs to make this relationship more clear, but I'd like to see the names themselves changes. I will propose a new naming convention via PR.

caveman99, 2023-07-25:
If we do that, we might want to wait till 2.2 or even 3.0 - last time we changes the presets all documentation hell broke loose.

charminULTRA, 2023-07-26:
> If we do that, we might want to wait till 2.2 or even 3.0 - last time we changes the presets all documentation hell broke loose.

Is it OK for me to submit a PR that we can build upon? Beyond the firmware settings and protobufs, I don't have a great idea of how many places in the overall project use the naming conventions, or where they are located.

JmCasler, 2023-07-26:
Changing the name is ecosystem wide. Not just on the firmware. It'll need to be synchronized across the board.

Advice is to leave this open and let this be the catch-all for the requests.

thebentern, 2023-10-10:
Closing and moving instead to item on the Meshtastic 3.0 project

closed 2023-10-10 by thebentern