meshtastic/firmware #2669: [Bug]: Serial data output to Console doesn't allow binary data

erict-powersoft, 2023-07-30:
### Category

Serial

### Hardware

T-Beam

### Firmware Version

1.2.19 beta

### Description

Line 252 in src/modules/SerialModule.cpp does

    serialPrint->printf("%s",` p.payload.bytes);

but should be

    serialPrint->write(p.payload.bytes, p.payload.size);

so that binary data is output completely.  At present a value of 0 in the payload will stop output.

I'm outputting serial data to the serial console after setting device.serial_enabled to false and override console_serial_port to true.  But since my serial messages are binary, the output was interrupted if a 0x00 was embedded in the serial message.

caveman99, 2023-07-31:
The simple text mode of the serial module is for exactly that. Simply ASCII Text. For transferring binary data, please use the Protobuf/API mode. Changing the way the textmessage mode works will break other applications that rely on the control characters embedded in the flow. For more reference: https://meshtastic.org/docs/development/device/client-api

erict-powersoft, 2023-07-31:
Is the following sentence still true then?
'SIMPLE' operate as an dumb UART tunnel. What goes in will come out, Requires a channel named 'serial'.
[mode definition in https://meshtastic.org/docs/settings/moduleconfig/serial]

GUVWAF, 2023-07-31:
@caveman99 He's using the `SIMPLE` mode, not `TEXTMSG`. Given that description, isn't using `write()` here as he proposes more correct?

mc-hamster, 2023-07-31:
> @caveman99 He's using the `SIMPLE` mode, not `TEXTMSG`. Given that description, isn't using `write()` here as he proposes more correct?

I think @erict-powersoft is correct here. This bug has been here for over two years! 

Reopening

caveman99, 2023-07-31:
yes, indeed i was confusing the 2 modes.

GUVWAF, 2023-07-31:
Seems it was corrected 7 months ago (https://github.com/meshtastic/firmware/commit/e1feb5eb30f67c06661f649a403d7d4ed6696c84), but afterwards changed again. Got a PR up to fix it again.

PR #2670 (2023-07-31): SerialModule SIMPLE mode: use `write()` instead of `printf()`
Fixes #2669.
closed 2023-07-31 by caveman99