meshtastic/firmware #2693: [Board]: WiPhone

Nikguy321, 2023-08-08:
### SOC

ESP32

### Lora IC

RFM95W

### Product Link

https://www.wiphone.io/

### Description

It's an esp32 based "phone" with a rfm95w, and has a built in lora messaging app with contacts and so on. Would be great if there was a meshtastic app

Nikguy321, 2023-08-08:
> Not a compatible license, if support is added for WiPhone it would overwrite their firmware.

I don't think that's a deal breaker if done well

gigawatts, 2023-08-08:
@caveman99 started this work: https://github.com/caveman99/Meshtastic-Wiphone

I can confirm it compiles and works (replacing WiPhone stock firmware entirely, of course), but there is currently no local UI or input functionality, only Bluetooth/WiFi.

I opened an issue (https://github.com/caveman99/Meshtastic-Wiphone/issues/2) on their repo to see if any further progress had been made, considering the T-Deck and T-Watch variants use the same display driver as the WiPhone.

I'm willing to test anything in progress!

closed 2023-10-29 by thebentern