meshtastic/firmware #2698: [Board]: nrf52820 custom board

Kralg, 2023-08-09:
### SOC

NRF52

### Lora IC

RFM95

### Product Link

Custom

### Description

Hello,
I would like to run meshtastic on a custom board with nrf52820 and RFM95. Does anyone can comment on the memory usage of nrf52 series firmware and should it work on nrf52820?

markbirss, 2023-08-09:
@Kralg already supported, 1mb flash and 256kb ram enough for nrf52

various official targets
t-echo, rak6431 including recent new hardware Nano G2 Ultra
https://github.com/meshtastic/firmware/issues/2658

have made various diy boards already 
check out my youtube
https://www.youtube.com/channel/UCeu8F_EsAyoSPVpm8v498oQ

https://meshtastic.discourse.group/t/wip-diy-nrf82540-nicerf-868-4-2-inch-epaper/5552
https://meshtastic.discourse.group/t/wip-new-nrf82540-device-makerpython-nrf82540/6029

Nano 33 BLE

busy with Makerdiary nRF52840 connect kit

Kralg, 2023-08-10:
Hi Markbirss,

Thanks for your email. Yeah so the plan would be to use nrf52820 which has lower memory: 256 KB Flash, 32 KB RAM. That is why the question is can anybody comment on the actual memory need and can it work on nrf52820 256 KB Flash, 32 KB RAM

markbirss, 2023-08-10:
@Kralg 

Ok, no 256 KB flash, 32 KB ram wont be enough for nrf52, but there is work on STM32 listed below

HARDWARE: NRF52840 64MHz, 243KB RAM, 796KB Flash

Linking .pio/build/nrf52840dk-geeksville/firmware.elf
Checking size .pio/build/nrf52840dk-geeksville/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  24.4% (used 60832 bytes from 248832 bytes)
Flash: [=====     ]  52.3% (used 426124 bytes from 815104 bytes)

Checking size .pio/build/t-echo/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  24.3% (used 60504 bytes from 248832 bytes)
Flash: [=====     ]  54.0% (used 440444 bytes from 815104 bytes)

PLATFORM: ST STM32 (15.6.0) > BB-STM32WL
HARDWARE: STM32WLE5CCU 48MHz, 64KB RAM, 256KB Flash

https://github.com/meshtastic/firmware/issues/2539

closed 2023-08-11 by thebentern

Kralg, 2023-09-01:
Hello,

Thank you for the answer. I have one more question, can you please write if the SX DIO0-DIO5 pins are crucial to be connected to MCU to run meshtastic on nrf or esp?