meshtastic/firmware #2702: [Bug]: Delivery status not working

MasterSPB, 2023-08-10:
### Category

Other

### Hardware

DIY

### Firmware Version

2.1.23

### Description

When using modem preset VERY LONG SLOW often got incorrect delivery status.
Test conditions:
- I have couple identical devices (distance - 1m)
- devices have identical settings (also frequency changed to custom 870MHz for clear test) 
I'm entry 3 messages with small delay - all messages delivered, but sender can't got ack about it.

This bug reproduces randomly. 
some time all messages has delivery mark, another time only one or two messages has delivery mark

[image]

GUVWAF, 2023-08-10:
Does it work correctly using modem preset LongFast? 

Which DIY hardware are you using? Which LoRa chip is it and does it have a TCXO? VeryLongSlow has a narrow bandwidth and that might require one.

MasterSPB, 2023-08-10:
LongFast - work good 
Dyi V1 (ESP32 + E22-900M30S)
no TCXO

GUVWAF, 2023-08-10:
@thebentern according to the RadioLib comment referenced [here](https://github.com/meshtastic/firmware/commit/d9ab6986ade1613f2b1d9cbc3fa2299d6074429b#diff-838a2688457a823341cda8ec32a3bb9bf7cf84fef97791b28344f30ac64d70b9R28), specifically for the E22-900M30S, the TCXO Voltage has to be set to 2.4, but in that commit it was changed again to 1.8V (which is probably correct for another E22 module). Maybe we have to set it differently for different modules in order for the TCXO to work correctly?

GUVWAF, 2023-08-10:
In hindsight I don't think it's a bug, the voltage should be okay for this module. @thebentern mentioned that it also doesn't work well with TCXO for this preset. Maybe the bandwidth is just too narrow for this hardware; I would advise to use LongSlow then. 
I don't have EByte modules to test with.

MasterSPB, 2023-08-11:
I checked with SlowLong - issue still remains

GUVWAF, 2023-08-11:
Can you try it with the devices more than 1 meter apart? For example in another room in order to ensure the receivers don't get overloaded.

MasterSPB, 2023-08-11:
ok, 5+ meters with walls - working correct

closed 2023-08-11 by thebentern