meshtastic/firmware #2704: [Bug]: Heltec Wireless Tracker screen doesnt work

GMart, 2023-08-10:
### Category

Hardware Compatibility

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.1.23

### Description

Wireless Tracker TFT screen doesn't work. I have board version 1.1

### Relevant log output

```Shell
From version 2.1.20: 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 3 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 3 Filesystem files (491520/1048576 Bytes):
DEBUG | ??:??:?? 3  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 4  /prefs/db.proto (235 Bytes)
DEBUG | ??:??:?? 4  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 4  /static/Logo_Black.svg.gz (592 Bytes)
DEBUG | ??:??:?? 4  /static/Logo_White.svg.gz (600 Bytes)
DEBUG | ??:??:?? 4  /static/apple-touch-icon.png.gz (3162 Bytes)
DEBUG | ??:??:?? 4  /static/favicon.ico.gz (2221 Bytes)
DEBUG | ??:??:?? 4  /static/icon.svg.gz (842 Bytes)
DEBUG | ??:??:?? 4  /static/index-a329437c.js.gz (202854 Bytes)
DEBUG | ??:??:?? 4  /static/index-ad0dcc03.css.gz (17092 Bytes)
DEBUG | ??:??:?? 4  /static/index-fc2e9253.js.gz (347 Bytes)
DEBUG | ??:??:?? 4  /static/index.html.gz (546 Bytes)
DEBUG | ??:??:?? 4  /static/maplibre-gl-d8126e41.js.gz (205728 Bytes)
DEBUG | ??:??:?? 4  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 4  /static/site.webmanifest.gz (196 Bytes)
DEBUG | ??:??:?? 4 Using analog input 1 for battery level
INFO  | ??:??:?? 4 ADCmod: ADC characterization based on default reference voltage
INFO  | ??:??:?? 4 Scanning for i2c devices...
DEBUG | ??:??:?? 4 Scanning for i2c devices on port 1
INFO  | ??:??:?? 130 No I2C devices found
DEBUG | ??:??:?? 130 acc_info = 0
INFO  | ??:??:?? 130 Meshtastic hwvendor=48, swver=2.1.20.470363d
DEBUG | ??:??:?? 130 Setting random seed 1601805042
DEBUG | ??:??:?? 130 Total heap: 260240
DEBUG | ??:??:?? 130 Free heap: 218436
DEBUG | ??:??:?? 130 Total PSRAM: 0
DEBUG | ??:??:?? 130 Free PSRAM: 0
DEBUG | ??:??:?? 130 NVS: UsedEntries 70, FreeEntries 560, AllEntries 630, NameSpaces 3
DEBUG | ??:??:?? 130 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 130 Number of Device Reboots: 3
DEBUG | ??:??:?? 130 OTA firmware version 0.2.1.2fbe7fc
INFO  | ??:??:?? 130 Initializing NodeDB
INFO  | ??:??:?? 130 Loading /prefs/db.proto
INFO  | ??:??:?? 130 Loaded saved devicestate version 20
INFO  | ??:??:?? 130 No /prefs/config.proto preferences found
INFO  | ??:??:?? 130 Installing default LocalConfig
INFO  | ??:??:?? 130 Setting default channel and radio preferences!
DEBUG | ??:??:?? 130 Expanding short PSK #1
INFO  | ??:??:?? 130 Wanted region 0, using UNSET
INFO  | ??:??:?? 130 No /prefs/module.proto preferences found
INFO  | ??:??:?? 130 Installing default ModuleConfig
INFO  | ??:??:?? 130 Loading /prefs/channels.proto
INFO  | ??:??:?? 130 Loaded saved channelFile version 20
INFO  | ??:??:?? 130 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 130 Number of Device Reboots: 3
[... 135 lines cut ...]
DEBUG | ??:??:?? 428 [Router] ESP32 crypt fr=fa4f80c4, num=2aebe53a, numBytes=21!
DEBUG | ??:??:?? 428 [Router] decoded message (id=0x2aebe53a fr=0xc4 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxSNR=6 rxRSSI=-46)
DEBUG | ??:??:?? 428 [Router] handleReceived(REMOTE) (id=0x2aebe53a fr=0xc4 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxSNR=6 rxRSSI=-46)
DEBUG | ??:??:?? 428 [Router] Module 'position' wantsPacket=1
INFO  | ??:??:?? 428 [Router] Received position from=0xfa4f80c4, id=0x2aebe53a, portnum=3, payloadlen=17
INFO  | ??:??:?? 428 [Router] POSITION node=fa4f80c4 l=17 LAT LON MSL TIME
DEBUG | 00:07:07 428 [Router] Upgrading time to RTC 1690945152 secs (quality 2)
DEBUG | 02:59:12 428 [Router] Read RTC time as 1690945152
INFO  | 02:59:12 428 [Router] updatePosition REMOTE node=0xfa4f80c4 time=1690945152, latI=412440000, lonI=-858500000
DEBUG | 02:59:12 428 [Router] Node status update: 2 online, 2 total
DEBUG | 02:59:12 428 [Router] Module 'position' considered
DEBUG | 02:59:12 428 [Router] Module 'routing' wantsPacket=1
INFO  | 02:59:12 428 [Router] Received routing from=0xfa4f80c4, id=0x2aebe53a, portnum=3, payloadlen=17
DEBUG | 02:59:12 428 [Router] Routing sniffing (id=0x2aebe53a fr=0xc4 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxSNR=6 rxRSSI=-46)
INFO  | 02:59:12 428 [Router] Rebroadcasting received floodmsg to neighbors
DEBUG | 02:59:12 428 [Router] Expanding short PSK #1
DEBUG | 02:59:12 428 [Router] Using AES128 key!
DEBUG | 02:59:12 428 [Router] ESP32 crypt fr=fa4f80c4, num=2aebe53a, numBytes=21!
WARN  | 02:59:12 428 [Router] send - lora tx disable because RegionCode_Unset
DEBUG | 02:59:12 428 [Router] Delivering rx packet (id=0x2aebe53a fr=0xc4 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxSNR=6 rxRSSI=-46)
DEBUG | 02:59:12 428 [Router] Update DB node 0xfa4f80c4, rx_time=0, channel=0
DEBUG | 02:59:12 428 [Router] Forwarding to phone (id=0x2aebe53a fr=0xc4 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxSNR=6 rxRSSI=-46)
DEBUG | 02:59:12 428 [Router] Module 'routing' considered
DEBUG | 02:59:15 432 [Power] Battery: usbPower=0, isCharging=0, batMv=0, batPct=0
```

lyusupov, 2023-08-12:
>  I have board version 1.1

http://community.heltec.cn/t/heltec-htit-tracker-v1-0/13364/10

Gary-in-Geneva, 2023-08-26:
I can confirm that a Heltec HTIT-Tracker V1.1 does not have a display using firmware V2.2.2f35c7be alpha. 

The Bluetooth LORA, and serial monitor are working.

caveman99, 2023-08-27:
I have a 1.0 board and wonder if there's a programmatic way to tell them both apart? I'd hate spinning different firmwares for both.

caveman99, 2023-08-27:
@GMart @Gary-in-Geneva can you mplease test the firmware from https://github.com/meshtastic/firmware/suites/15529928794/artifacts/886460425 ?

Gary-in-Geneva, 2023-08-28:
I erased and flashed both firmware-heltec-wireless-tracker-2.2.4.bc509e0.bin and the firmware-heltec-wireless-tracker-2.2.4.bc509e0-update.bin. There was no display.

caveman99, 2023-08-28:
Dang. My 1.0 still works with the detect code. Back to the drawing board.

Gary-in-Geneva, 2023-08-28:
I was able to get the Ver 1.1 to display using the Arduino IDE by making the following changes:

C:\Users\{user}\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.8\libraries\DISPLAY\src\HT_st7735.h
/*** Redefine if necessary ***/
#define ST7735_CS_Pin        38
#define ST7735_REST_Pin      39
#define ST7735_DC_Pin        40
#define ST7735_SCLK_Pin      41
#define ST7735_MOSI_Pin      42
#define ST7735_LED_K_Pin     21
#define ST7735_VTFT_CTRL_Pin -1

C:\Users\{user}\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.8\variants\wireless_track\pins_arduino.h
Line 74 should read as follows:

static const uint8_t Vext = 3;

In the program so the display would be visible:

void VextON(void)
{
  pinMode(Vext,OUTPUT);
  digitalWrite(Vext, HIGH);   // active HIGH
}

Hope it helps. It is difficult debugging without having the hardware to test with. Warning: I am just a hack - use the above information with caution :-)

GMart, 2023-08-28:
Yes, that is what I found looking at the schematics. VExt has to be redefined and the TFT backlight has to be active high. I just haven't had time to compile the code and flash it. We may need a different board version defined, as it's reverse of the V1.0.

Szetya, 2023-09-05:
In variants/heltec_wireless_tracker/pins_arduino.h changed from Vext=36 to Vext=3
variant.h > changed from #define ST7735_BL 45 to #define ST7735_BL 21 and #define ST7735_BACKLIGHT_EN 21 and added: #define VEXT_INVERTED 1
In main.cpp >
#ifdef VEXT_ENABLE
    pinMode(VEXT_ENABLE, OUTPUT);
    #ifndef VEXT_INVERTED
    digitalWrite(VEXT_ENABLE, 0); // turn on the display power v1.0
    #else
    digitalWrite(VEXT_ENABLE, 1); //v1.1 display power
    #endif
#endif

it works perfectly. :)
[image]
[image]
[image]

cherwey, 2023-09-15:
> In variants/heltec_wireless_tracker/pins_arduino.h changed from Vext=36 to Vext=3 variant.h > changed from #define ST7735_BL 45 to #define ST7735_BL 21 and #define ST7735_BACKLIGHT_EN 21 and added: [...]

is good work!!!, by the way, the battery display 0 in new board... is't something changed??

Szetya, 2023-09-15:
> is good work!!!, by the way, the battery display 0 in new board... is't something changed??

The second picture at the top shows the wiring diagram of the two versions. I didn't even notice the battery voltage. Although I have only tested it from USB.

cherwey, 2023-09-15:
here is the res  https://resource.heltec.cn/download/Wireless_Tracker

cherwey, 2023-09-15:
wireless tracker has low power circuit in 1.0 and 1.1(1.1 not support vgnss_ctrl and vtft_ctrl). so we get vbat must change gpIO2(ADC_Ctrl)

Szetya, 2023-09-16:
> wireless tracker has low power circuit in 1.0 and 1.1(1.1 not support vgnss_ctrl and vtft_ctrl). so we get vbat must change gpIO2(ADC_Ctrl)

The GPIO2 leg must be high level for the ADC (GPIO1) to measure. This is so that the battery is only on the voltage divider for the duration of the measurement due to power consumption.
An option is to configure one of the LED, Buzzer, Vibra to GPIO2 in the External Notification Config section of the device settings menu. Not pretty but it works. :) :D

cherwey, 2023-09-16:
I used     pinMode((gpio_num_t)ADC_CTRL, OUTPUT);    digitalWrite((gpio_num_t)ADC_CTRL, HIGH);, but is not work . I get raw= 4095, v in gpio change to  3.2V...

Szetya, 2023-09-16:
> I used pinMode((gpio_num_t)ADC_CTRL, OUTPUT); digitalWrite((gpio_num_t)ADC_CTRL, HIGH);, but is not work . I get raw= 4095, v in gpio change to 3.2V...

Have you declared the ADC_CTRL variable? I solved it very simply.
    pinMode(2, OUTPUT);
    digitalWrite(2, 1);
[image]

    :D

cherwey, 2023-09-16:
yes, I had #define ADC_CTRL [ADC1_GPIO2_CHANNEL, did i something wrong?
[image]
[image]

cherwey, 2023-09-16:
yes, u are right ,i  type ADC1_CHANNEL_1_GPIO_NUM(2) to ADC1_GPIO2_CHANNEL(1), it work now..... ：P

cherwey, 2023-09-16:
[image]

cherwey, 2023-09-16:
[image]

cherwey, 2023-09-17:
> > I used pinMode((gpio_num_t)ADC_CTRL, OUTPUT); digitalWrite((gpio_num_t)ADC_CTRL, HIGH);, but is not work . I get raw= 4095, v in gpio change to 3.2V...
> 
> Have you declared the ADC_CTRL variable? I solved it very simply. pinMode(2, OUTPUT); digitalWrite(2, 1); [image]
> 
> ```
> :D 
> ```

find new bug, the screen never sleep.

cherwey, 2023-09-17:
in sleep.cpp : 
[image]

caveman99, 2023-09-18:
Please check the firmware from that latest build. It should work on 1.0 and 1.1

swiftarf, 2023-10-04:
I've wiped and installed all v 2.2.x firmwares on my 1.1 tracker and have had no display either.

swiftarf, 2023-10-04:
Well, I've had success with firmware version 2.2.11.b6b2c635

jp-bennett, 2023-10-04:
> Well, I've had success with firmware version 2.2.11.b6b2c635

It needs a bit more sanity checking, but a dev finally got the 1.1 board in hand to work on

closed 2023-10-05 by jp-bennett

Gary-in-Geneva, 2023-10-13:
Flashed 2.2.11 on 4 different Trackers V1.1 - all have a display! Good work!

commit bc509e08 (2023-09-18): Possibly fix #2704
commit fc06754e (2023-10-05): Possibly fix #2704 "Heltec Wireless Tracker screen doesn't display anything" (#2749)

* fix #2704