meshtastic/firmware #2706: [Bug]: Heltec ver 2 hardware with GPS - Ver 2.2.0 Error 3

Gary-in-Geneva, 2023-08-10:
### Category

Hardware Compatibility

### Hardware

Heltec v2

### Firmware Version

2.2.0  alpha

### Description

Using Web flasher to wipe and install firmware (Nice tool!)
Ver 2.1.23 beta loads and appears to be working 

Using Web flasher to wipe and install firmware
Upon restarting Ver 2.2.0 Error 3 - no radio. I verified this on 3 different heltec ver 2 devices. When attempting to do blue tooth pairing this version firmware also causes the Android app to indicate that it needs to be updated. All apps and the phone are up to date

caveman99, 2023-08-14:
Please check if there's still an issue with 2.2.1

Gary-in-Geneva, 2023-08-26:
Sorry for the late reply. It is working with 2.2.1 and 2.23

closed 2023-08-26 by Gary-in-Geneva