meshtastic/firmware #2707: [Board]: chatter 2.0 by circuitmess

Nikguy321, 2023-08-10:
### SOC

ESP32

### Lora IC

Not sure guessing rfm95w but not sure.

### Product Link

https://circuitmess.com/pages/chatter-2-0

### Description

I know a few people (and some online) that start with this device for lora messaging before upgrading. I'm sure a few ppl have a few of these esp32 based kits laying around that they would like to add to the mesh. I don't own one myself.

jp-bennett, 2023-08-10:
We looked into this device not too terribly long ago. It looks like it was a special Christmas product sold exclusively through Sam's Club. If Circuitmess confirmed that they were going to make this available on an ongoing basis, developers would be a lot more interested in porting.

caveman99, 2023-08-25:
Sorry, closing this. The device is EOL and the developer engery is better invested in new devices.

closed 2023-08-25 by caveman99