meshtastic/firmware #2717: [Bug]: Firmware 2.2.0 drops fixed position

mrekin, 2023-08-13:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.2.0.9f6584b

### Description

Node based on Heltec Wireless stick.
1. Updated from 2.1.5 to 2.2.0 using web flasher.
2. Early saved settings imported in andriod app.
3. Configured fixed position in android app (2.2.0 version)
4. After little time found that fixed position is 0.0.
5. Configured fixed position again.
6. After ~20 min fixed position dropped again.

With 2.15 firmware all works as expected.

mrekin, 2023-08-13:
Firmware 2.2.1 the same - position drops after node reboot

thebentern, 2023-08-19:
@mrekin can you try out the 2.2.2 release?

mrekin, 2023-08-20:
@thebentern yes, i'll test in several hours

mrekin, 2023-08-21:
Node updated to 2.2.2 (no erase), wifi enabled.
Android app 2.2.2
1. Set  Lat/Long/Alt in android app
2. Check if coordinates displayed in node info
3. Disconnect- connect to node: coordinates still here
4. **Power off - power on : no coordinates in node info.**

Now repeat 1-3 step and will wait for 1-2 hours to check if coordinates still here.

mrekin, 2023-08-21:
4 hours testing - position info not dropped.
So problem exist only if node reboots/power off

RUNNER2k23, 2023-08-23:
Same for T-Beam on 2.2.2.f35c7be, after a reboot the fixed position ist reset

thebentern, 2023-08-24:
Can you post your full position config output from meshtastic --info?

mc-hamster, 2023-08-24:
> Can you post your full position config output from meshtastic --info?

It's 4:20am, you're awake?

thebentern, 2023-08-24:
> > Can you post your full position config output from meshtastic --info?
> 
> It's 4:20am, you're awake?

It's 6:20 here in CDT. I've been up since 5:45. ;-)

mc-hamster, 2023-08-24:
> > > Can you post your full position config output from meshtastic --info?
> > 
> > 
> > It's 4:20am, you're awake?
> 
> It's 6:20 here in CDT. I've been up since 5:45. ;-)

It's 4:20am, I'm awake?

thebentern, 2023-08-24:
@mrekin I believe I have a solution for this issue. It will go out in the next firmware release.

thebentern, 2023-08-27:
Fixed in 2.2.3 release

closed 2023-08-27 by thebentern

mrekin, 2023-08-28:
@thebentern 2.2.3 still drops position.
Found that I need turn off 'smart GPS' and 'GPS enabled' to keep fixed coordinates, so i'll use this config.

garthvh, 2023-08-28:
> @thebentern 2.2.3 still drops position.
> Found that I need turn off 'smart GPS' and 'GPS enabled' to keep fixed coordinates, so i'll use this config.

Fixed position is just the last position the device has, if you turn on the GPS it will quickly override your fixed position value. Turning off smart position is doing nothing for you.

mrekin, 2023-08-29:
> Fixed position is just the last position the device has, if you turn on the GPS it will quickly override your fixed position value.

Fixed position is just manually entered values in my case, because  hw doesn't have GPS module.
As user I expect to get clear behavior :)  If I set fixed position - I expect it will work or return error\exception. Maybe this is app bug, but firmware allow set both parameters at the same time and provides unexpected behavior