meshtastic/firmware #2721: [Bug]: Neighbor `lastRxTime` and `nodeBroadcastIntervalSecs` fields are not set in firmware

ajmcquilkin, 2023-08-14:
### Category

Other

### Hardware

T-Beam, Rak4631

### Firmware Version

2.2.0.9f6584b Alpha

### Description

When my node receives incoming `NeighborInfo` packets, the fields `neighborInfoPacket.neighbors[id].lastRxTime` and `neighborInfoPacket.neighbors[id].nodeBroadcastIntervalSecs` are not being set in firmware and are always received as 0. An example of this is shown below (pulled from MNMC state debug pane).

### Relevant log output

```Shell
{
  "4": {
    "data": {
      "lastSentById": 4,
      "neighbors": [
        {
          "lastRxTime": 0,
          "nodeBroadcastIntervalSecs": 0,
          "nodeId": 3484259578,
          "snr": 6
        },
        {
          "lastRxTime": 0,
          "nodeBroadcastIntervalSecs": 0,
          "nodeId": 2438984268,
          "snr": 6
        }
      ],
      "nodeBroadcastIntervalSecs": 60,
      "nodeId": 4
    },
    "packet": {
      "channel": 0,
      "delayed": 0,
      "from": 4,
      "hopLimit": 3,
      "id": 506686815,
      "payloadVariant": {
        "decoded": {
          "dest": 0,
          "emoji": 0,
          "payload": [
            8,
            4,
            16,
            4,
            24,
            60,
            34,
            11,
            8,
            250,
            169,
            182,
            253,
            12,
            21,
            0,
            0,
            192,
            64,
            34,
            11,
            8,
            204,
            228,
            255,
            138,
            9,
            21,
            0,
            0,
            192,
            64
          ],
          "portnum": 71,
          "replyId": 0,
          "requestId": 0,
          "source": 0,
          "wantResponse": false
        }
      },
      "priority": 64,
      "rxRssi": 0,
      "rxSnr": 0,
      "rxTime": 1691963808,
      "to": 4294967295,
      "wantAck": false
    }
  },
  "2438984268": {
    "data": {
      "lastSentById": 2438984268,
      "neighbors": [
        {
          "lastRxTime": 0,
          "nodeBroadcastIntervalSecs": 0,
          "nodeId": 4,
          "snr": 7
        },
        {
          "lastRxTime": 0,
          "nodeBroadcastIntervalSecs": 0,
          "nodeId": 3484259578,
          "snr": 5
        }
      ],
      "nodeBroadcastIntervalSecs": 60,
      "nodeId": 2438984268
    },
    "packet": {
      "channel": 0,
      "delayed": 0,
      "from": 2438984268,
      "hopLimit": 3,
      "id": 714301993,
      "payloadVariant": {
        "decoded": {
          "dest": 0,
          "emoji": 0,
          "payload": [
            8,
            204,
            228,
            255,
            138,
            9,
            16,
            204,
            228,
            255,
            138,
            9,
            24,
            60,
            34,
            7,
            8,
            4,
            21,
            0,
            0,
            224,
            64,
            34,
            11,
            8,
            250,
            169,
            182,
            253,
            12,
            21,
            0,
            0,
            160,
            64
          ],
          "portnum": 71,
          "replyId": 0,
          "requestId": 0,
          "source": 0,
          "wantResponse": false
        }
      },
      "priority": 0,
      "rxRssi": -42,
      "rxSnr": 6.25,
      "rxTime": 1691963784,
      "to": 4294967295,
      "wantAck": false
    }
  },
  "3484259578": {
    "data": {
      "lastSentById": 3484259578,
      "neighbors": [
        {
          "lastRxTime": 0,
          "nodeBroadcastIntervalSecs": 0,
          "nodeId": 4,
          "snr": 6
        },
        {
          "lastRxTime": 0,
          "nodeBroadcastIntervalSecs": 0,
          "nodeId": 2438984268,
          "snr": 6
        }
      ],
      "nodeBroadcastIntervalSecs": 60,
      "nodeId": 3484259578
    },
    "packet": {
      "channel": 0,
      "delayed": 0,
      "from": 3484259578,
      "hopLimit": 3,
      "id": 639718602,
      "payloadVariant": {
        "decoded": {
          "dest": 0,
          "emoji": 0,
          "payload": [
            8,
            250,
            169,
            182,
            253,
            12,
            16,
            250,
            169,
            182,
            253,
            12,
            24,
            60,
            34,
            7,
            8,
            4,
            21,
            0,
            0,
            192,
            64,
            34,
            11,
            8,
            204,
            228,
            255,
            138,
            9,
            21,
            0,
            0,
            192,
            64
          ],
          "portnum": 71,
          "replyId": 0,
          "requestId": 0,
          "source": 0,
          "wantResponse": false
        }
      },
      "priority": 0,
      "rxRssi": -20,
      "rxSnr": 6.25,
      "rxTime": 1691963828,
      "to": 4294967295,
      "wantAck": false
    }
  }
}
```

GUVWAF, 2023-08-14:
This is because these values are not send over the mesh (to save 8 bytes per neighbor), and the client only gets a copy of this. However, you can source the `broadcast_interval_secs` of other nodes when you get their NeighborInfo packet. 
We could also add these values to the copy that is sent to the client, but it would then only be for the neighbors of the node connected to the client, because we don't have these values for 'neighbors of neighbors'. 
Do you need the `last_rx_time`? The nodes will already remove neighbors based on this, so you can remove edges when a new NeighborInfo packet doesn't contain these neighbors anymore, or when you didn't receive a NeighborInfo packet from that node for a while.

ajmcquilkin, 2023-08-14:
Ah didn't realize this was a data-saving thing from the device. I don't specifically need the `last_rx_time` since the client just assumes that the device immediately forwards the packet.

The client currently uses the `neighbors` field to add nodes to the graph if they're neighbors of a node that has sent a `NeighborInfo` packet, which is why I was wondering about the `broadcast_interval_secs`. It wouldn't be too hard to only add nodes to the graph if the client has received a neighbor info packet directly from them, although it might increase the latency of the graph updates.

Thanks for the clarification, happy to close the issue on that note.

closed 2023-08-14 by ajmcquilkin