meshtastic/firmware #2734: [Bug]: T-Echo wrong time

RUNNER2k23, 2023-08-20:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.1.11.5ec624d

### Description

Time from PCF8563_RTC is incorrect and seems to not be updated from GPS time.

As seen in the logs no matter if GPS time is year 2009 or 2023:
- RTC time is set to 536939680
- RTC time is read as 536901136
- time HH:MM:SS is always correct, but date is not
 
Maybe related: https://github.com/meshtastic/firmware/pull/1729

### Relevant log output

```Shell
DEBUG | 12:48:07 14 [GPS] NMEA GPS time 2009-03-27 12:48:07
DEBUG | 12:48:07 14 [GPS] Upgrading time to RTC 0 secs (quality 1238158087)
DEBUG | 12:48:07 14 [GPS] PCF8563_RTC setDateTime 2009-03-27 12:48:07 536939680
DEBUG | 12:48:07 14 [GPS] Read RTC time from PCF8563 as 536901136
DEBUG | 12:48:07 14 [GPS] publishing pos@0:2, hasVal=0, GPSlock=0
DEBUG | 12:48:07 14 [GPS] onGPSChanged() pos@0, time=1238158087, lat=0, lon=0, alt=0
INFO  | 12:48:07 14 [GPS] updatePosition LOCAL pos@0, time=1238158087, latI=0, lonI=0, alt=0

After reboot when the device had a GPS fix:
DEBUG | 12:54:41 13 [GPS] NMEA GPS time 2023-08-18 12:54:12
DEBUG | 12:54:41 13 [GPS] Upgrading time to RTC 0 secs (quality 1692363252)
DEBUG | 12:54:12 13 [GPS] PCF8563_RTC setDateTime 2023-08-18 12:54:12 536939680
DEBUG | 12:54:12 13 [GPS] Read RTC time from PCF8563 as 536901136
DEBUG | 12:54:12 13 [GPS] publishing pos@0:2, hasVal=0, GPSlock=0
DEBUG | 12:54:12 13 [GPS] onGPSChanged() pos@0, time=1692363252, lat=0, lon=0, alt=0
INFO  | 12:54:12 13 [GPS] updatePosition LOCAL pos@0, time=1692363252, latI=0, lonI=0, alt=0
```

caveman99, 2023-08-20:
Can you please re-test with a 2.2. series firmware? also make soure your phone time is correct, in case this is connected.

RUNNER2k23, 2023-08-20:
Time on phone is set correctly (from network, GMT+2).
Tested again with current firmware (I uploaded Meshtastic_nRF52_factory_erase.uf2 before update):

```
INFO  | ??:??:?? 4 Meshtastic hwvendor=7, swver=2.2.2.f35c7be
[...]
DEBUG | ??:??:?? 4 Read RTC time from PCF8563 as 536901444
DEBUG | 18:03:28 4 Using MSL altitude model
[...]
DEBUG | 18:03:34 11 [GPS] NMEA GPS time 2009-03-29 18:03:00
DEBUG | 18:03:34 11 [GPS] Upgrading time to RTC 0 secs (quality 1238349780)
DEBUG | 18:03:00 11 [GPS] PCF8563_RTC setDateTime 2009-03-29 18:03:00 0
DEBUG | 18:03:00 11 [GPS] Read RTC time from PCF8563 as 536901444
DEBUG | 18:03:00 11 [GPS] publishing pos@0:2, hasVal=0, GPSlock=0
DEBUG | 18:03:00 11 [GPS] No GPS lock
DEBUG | 18:03:00 11 [GPS] onGPSChanged() pos@0, time=1238349780, lat=0, lon=0, alt=0
INFO  | 18:03:00 11 [GPS] updatePosition LOCAL pos@0, time=1238349780, latI=0, lonI=0, alt=0
```
Reboot after GPS fix:
```
DEBUG | ??:??:?? 3 Read RTC time from PCF8563 as 536901444
DEBUG | 18:11:18 3 Using MSL altitude model
[...]
DEBUG | 18:11:24 9 [GPS] NMEA GPS time 2023-08-20 18:11:22
DEBUG | 18:11:24 9 [GPS] Upgrading time to RTC 0 secs (quality 1692555082)
DEBUG | 18:11:22 9 [GPS] PCF8563_RTC setDateTime 2023-08-20 18:11:22 536953592
DEBUG | 18:11:22 9 [GPS] Read RTC time from PCF8563 as 536901444
DEBUG | 18:11:22 9 [GPS] publishing pos@0:2, hasVal=0, GPSlock=0
DEBUG | 18:11:22 9 [GPS] No GPS lock
DEBUG | 18:11:22 9 [GPS] onGPSChanged() pos@0, time=1692555082, lat=0, lon=0, alt=0
INFO  | 18:11:22 9 [GPS] updatePosition LOCAL pos@0, time=1692555082, latI=0, lonI=0, alt=0
```

Maybe interesting:
- read RTC time has changed from 536901136 (2.1.11) to 536901444 (2.2.2) (+308)
- setDateTime after GPS fix has changed from 536939680 (2.1.11) to 536953592 (2.2.2) (+13912)
- setDateTime before GPS fix is now 0 (2.2.2)
- time offset before/after GPS fix is constant: 5257 days / 454 204 800 seconds

RUNNER2k23, 2023-08-20:
Maybe also relevant:

In this comment https://github.com/meshtastic/firmware/pull/1729#issuecomment-1266871745 the log output was:
`11:43:21 59 [GPS] Upgrading time to RTC 1664883802 secs (quality 0)`
Now:
`DEBUG | 18:11:24 9 [GPS] Upgrading time to RTC 0 secs (quality 1692555082)`
Time/quality swapped?

daniel-hofer, 2023-08-21:
I can confirm this issue with Firmware 2.2.2 on the T-Echo. Made a full reset and flashed the reset firmware before. Phone Settings are correct. Debug View shows at rx_time: 1557131585 now at 2023-08-21 10:33.
Had troubles to accomplish a GPS-Fix - switched the GPS Settings smart of and gps enabled from on to off and again to off. After the reboot a position, but lost it again some minutes after.

daniel-hofer, 2023-08-21:
[2.2.2.techo.txt](https://github.com/meshtastic/firmware/files/12395342/2.2.2.techo.txt)

daniel-hofer, 2023-08-23:
I updated to 2.2.2 and waited some hours with gps in range. The time was correct but the Date again 08.05.2019.
[techoraspberryterminal.txt](https://github.com/meshtastic/firmware/files/12423415/techoraspberryterminal.txt)
I wanted to look at the serial output and restarted the device and looked at the terminal and after some time, the epoch time was correct again! Feel free to analyze the attached Terminal output. Maybe there is something wrong.

daniel-hofer, 2023-08-31:
After flashing the firmware softrf and waiting for gps the date got corrected. Flashted meshtastic 2.2.3 after that without the resetting uf2 and since now the date/time-stamp works like a charm for many days.

closed 2023-09-27 by thebentern