meshtastic/firmware #2735: [Bug]: Problem with compile for linux-arm.

mate-dev, 2023-08-20:
### Category

Other

### Hardware

Other

### Firmware Version

2.2.3

### Description

There is a problem when compiling firmware for linux-arm.

### Relevant log output

```Shell
Compiling .pio/build/linux-arm/src/platform/portduino/PortduinoGlue.cpp.o
In file included from src/platform/portduino/PortduinoGlue.cpp:10:
/home/runner/.platformio/packages/framework-portduino/cores/portduino/linux/gpio/LinuxGPIOPin.h:15:10: fatal error: gpiod.h: No such file or directory

***************************************************************
* Looking for gpiod.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:gpiod.h"
* Web  > https://registry.platformio.org/search?q=header:gpiod.h
*
***************************************************************

   15 | #include <gpiod.h>
      |          ^~~~~~~~~
compilation terminated.
*** [.pio/build/linux-arm/src/platform/portduino/PortduinoGlue.cpp.o] Error 1
========================= [FAILED] Took 71.05 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
linux-arm      FAILED    00:01:11.047
==================== 1 failed, 0 succeeded in 00:01:11.047 ====================
Error: Process completed with exit code 1.
```

caveman99, 2023-08-21:
couple of things: 
- linux support is Work In Progress (read: it doesn't work). Please only try to use this if you wanna help make it work
- please use the raspi-portduino branch of this repository
- to go around the missing header file (it's not really a bug) install libgpiod-dev

Closing this, as it's not a bug as stated. Support is simply not finished.

closed 2023-08-21 by caveman99

irongiant33, 2024-01-23:
I found this when trying to get Linux Native working on Ubuntu 22.04.3 LTS and this fix sort of helped me, but what really helped me was running the following:

```
sudo apt-get -y install wget python3 g++ python3-venv vim ca-certificates libgpiod-dev libyaml-cpp-dev libbluetooth-dev
```

This was in your guys' Dockerfile (https://github.com/meshtastic/firmware/blob/4ae5443c3baaa10ac9368ecd53db104bde9936f7/Dockerfile#L14)

mate-dev, 2024-01-24:
@irongiant33 If you want working linux native for amd64, you can visit my repo https://github.com/mate-dev/mmrelaynode and download new version from releases. And here https://github.com/mate-dev/mmrelaynode/blob/master/.github/workflows/main.yml you can check how I build firmware in github workflows. By the way, I recommend reading the entire repo :)

irongiant33, 2024-01-24:
@mate-dev interesting, I'll check it out. Thanks!

mate-dev, 2024-01-25:
@irongiant33 I only recently found the Dockerfile that is here, I don't remember what I was based on when writing it.
Although something probably changed with node.js versions and I can't compile the latest image, but in version 2.2.18 it was ok. Tomorrow maybe I will look at it.