meshtastic/firmware #2736: [Bug]: T-Echo Button to switch display view not working

daniel-hofer, 2023-08-21:
### Category

Hardware Compatibility

### Hardware

T-Echo

### Firmware Version

2.2.2.f35c7be

### Description

Since 2.2.x the Button on the left upper side of the T-Echo does nothing with the screen but the red LED blinks after pressing the button.
So I cannot switch the page of the display

caveman99, 2023-08-21:
[image]

Are you sure you mean the UPPER left button? That's reset. LOWER left is the user button.

daniel-hofer, 2023-08-21:
> [image]
> 
> Are you sure you mean the UPPER left button? That's reset. LOWER left is the user button.

Oh no, sorry about that, my mistake! I got the keys mixed up because my other Node, a heltec v.3, has them the other way around. Sorry! I'll print myself a label for the T-Echo.

closed 2023-08-21 by daniel-hofer