meshtastic/firmware #2743: 🚀 [Feature Request]:

SidedFreedom, 2023-08-23:
### OS

iOS

### Description

I would like to send bitcoin raw transactions and ecash tokens over Lora. These can be very long texts. 

Could there be a feature to toggle on/off to send buffered messages through Lora with _special symbols denoting_ the beginning, addition and end of the buffered text so I could setup rules in python to parse the incoming texts correctly?

**For example this:**

YWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMw

**might become this:**

5StringsBegin (some language/symbols to denote incoming size)
1==vWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlM+

2+V0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem+

3+5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrY+

4+WVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLan+

5+JxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMwYWxrYWVMS0pPRWlMV0VLanJxOTI4M2Vmem5jcTNyMDlxajMw==

### Participation

- [ ] I am willing to submit a pull request for this issue.

garthvh, 2023-08-23:
Seems like you can do this already with multiple messages, what you have already looks to solve this.

closed 2023-08-23 by garthvh

SidedFreedom, 2023-08-23:
Thanks for the reply. Yes, I can do that, but I don't expect others to be able to self-buffer. I was just giving one use case, but buffering long messages would in my opinion be a good thing from the iOS side - which is something that I am not able to customize on my end of things unlike the MCU code.

garthvh, 2023-08-23:
Seems hard on the mesh and pretty confusing to have in the mobile apps.

caveman99, 2023-08-24:
Your best bet if you wanna do this is to make a bespoke commandline or GUI client with the available API's. JS, Python, HTTP all available.