meshtastic/firmware #2747: [Feature Request]: Option to not notify on External Buzzer receiving MSG when phone app is connected on Bluetooth.

cordelster, 2023-08-24:
### Platform

ESP32

### Description

Would be excellent to have the option inn ExternalNotificationModule.cpp under line 371: moduleConfig.external_notification.alert_message_buzzer to add something in the way of if(meshtasitic_Bluetooth_ConnectionStatus) is connected....

Really rough example:
`            if (moduleConfig.external_notification.alert_message_buzzer) {
                LOG_INFO("externalNotificationModule - Notification Module (Buzzer)\n");
                if (_meshtastic_BluetoothConnectionStatus) {
                    LOG_INFO("externalNotificationModule - Notification Module (Buzzer disabled by BlueTooth)\n");
                } else {
                    isNagging = true;
                    if (!moduleConfig.external_notification.use_pwm) {
                        setExternalOn(2);
                    } else {
                        rtttl::begin(config.device.buzzer_gpio, rtttlConfig.ringtone);
                    }
                    if (moduleConfig.external_notification.nag_timeout) {
                        nagCycleCutoff = millis() + moduleConfig.external_notification.nag_timeout * 1000;
                    } else {
                        nagCycleCutoff = millis() + moduleConfig.external_notification.output_ms;
                    }
                }
            }`

I know just enough about C++ to make a mess, though I am still attempting to get this to work this week or next.

My use case is to alert residents of wild fire, which we are using the Bell for. But when someones phone app is connected the messaging tones from both devices is a little much, though if the phone app is not connected, the device should be smart enough to resume  alerts for messages. I'm  just not sure how to pull from that Class and turn it into a boolean. Still trying.

caveman99, 2023-08-24:
My 2 cents from being in the alerting/mobilization business for the better part of the last 25 years... The more noise it makes, the better. I'd vote for the 'KISS' approach we're using right now and not mod that. Every conditonal introduces new bug potential, if you want to make sure people survice, make sure they hear the alert.

closed 2023-08-24 by thebentern

thebentern, 2023-08-24:
Ext. Notification module config has 14 existing settings. I think we should be very cautious about adding any additional settings, as it's already a confusing one to configure from a user perspective.

cordelster, 2023-08-25:
Okay I'll work out the code, but several people have asked me to install a spst switch, and anything like my  iPhone, I  forget to turn it back on. I thought this was a better approach as a bell could still be sent regardless of connection status which is what we use for emergency paging/notification. Or I have shown a few how to turn off external message alerts in their app. We now have already 10 Repeater/routers up and I just ordered more. 53 households participating and growing.