meshtastic/firmware #2751: [Bug]:

Jeripimo, 2023-08-28:
### Category

Other

### Hardware

T-Echo

### Firmware Version

1.2.6

### Description

T-Echo originally loaded with softrf firmware. Working well.

After flashing firmware T-Echo_Meshtastic-1.2.64.fc48fcd.uf2 which is working and the afther that back to softrf GNSS states - at startup screen. 

After reading and  searchin the internet I read it could possibly be cause by change of L76 gps buad rate.

How can I set the baudrate back to 9600?

Jeripimo, 2023-08-28:
Just installed 2.2.3 but it says "no GPS module"  so that's not working for me.

How to reset the gps only

GUVWAF, 2023-09-01:
@jp-bennett is working on recovering connecting to a GPS module that has a different baud rate in #2752. Maybe you could try the firmware from the artifacts of that PR.

jp-bennett, 2023-09-01:
I haven't had an L76K to test it on. I think it will detect the GPS chip, but probably won't know how to reset the baud yet. But I'd love to get the log output from trying it.

Jeripimo, 2023-09-01:
> Oof. This one was really slow lol. It's not showing it found it and it says "No GPS Module" on the screen.
> DEBUG | 19:05:07 0 Probing for GPS at 9600
> WARN  | 19:05:09 2 Failed to find UBlox & MTK GNSS Module using baudrate 9600
> DEBUG | 19:05:09 2 Probing for GPS at 4800
> WARN  | 19:05:11 5 Failed to find UBlox & MTK GNSS Module using baudrate 4800
> DEBUG | 19:05:11 5 Probing for GPS at 38400
> WARN  | 19:05:13 7 Failed to find UBlox & MTK GNSS Module using baudrate 38400
> DEBUG | 19:05:13 7 Probing for GPS at 57600
> WARN  | 19:05:15 9 Failed to find UBlox & MTK GNSS Module using baudrate 57600
> DEBUG | 19:05:15 9 Probing for GPS at 115200
> WARN  | 19:05:17 11 Failed to find UBlox & MTK GNSS Module using baudrate 115200
> DEBUG | 19:05:17 11 GxGSA NOT available```

Isn't it a L76 gps module in tje T-Echo?

Jeripimo, 2023-09-01:
https://www.google.com/search?q=l76k+gps+module+reset+baudrate&oq=l76+&gs_lcrp=EgZjaHJvbWUqBggDEEUYOzIGCAAQRRg5MgYIARBFGEEyBggCEEUYOzIGCAMQRRg7MgYIBBBFGDsyBwgFEAAYgAQyBwgGEAAYgAQyBwgHEAAYgAQyBwgIEAAYgAQyBwgJEAAYgATSAQgzNjQwajBqNKgCALACAA&client=ms-unknown&sourceid=chrome-mobile&ie=UTF-8.    Does this help? I am a noob programming

jp-bennett, 2023-09-01:
I think I see, maybe, what's going on. I'll amend the PR, and we'll see if it makes a difference.

jp-bennett, 2023-09-03:
We've done quite a bit of work on this. Feel free to test again from https://github.com/meshtastic/firmware/actions/runs/6066814336#artifacts

I have a t-echo eventually coming in for further testing, so we'll eventually squash this issue for good.

Jeripimo, 2023-09-03:
I tried this one: [debug-elfs-2.2.4.7ad94da.zip](https://github.com/meshtastic/firmware/suites/15781704234/artifacts/900626632)
which is working in the t-echo. GPS is working and receiving satellites.

The I flashed the softrf firmware SoftRF-firmware-v1.3-nRF52.uf2  and a miracle  happened. 
Startup screen shows GNSS +  and gps is working again with softrf

Many thanks !!

closed 2023-09-03 by jp-bennett