meshtastic/firmware #2760: [Feature Request]: Add Philippines (PH) - AS923-3 (915-918 Mhz) LoRa Region Setting

markbirss, 2023-09-03:
### Platform

ESP32

### Description

There is currently no region setting for Philippines (PH) using AS923-3 (915-918 Mhz, 25mW)

https://www.reddit.com/r/HeliumNetwork/comments/m8ljdd/anyone_running_a_hotspot_in_the_philippines/
The 434/868 Mhz frequencies are not free
[image]

https://lora-alliance.org/wp-content/uploads/2021/05/RP-2-1.0.3.pdf
Page 16 of 94 

[image]

andrekir, 2023-09-03:
Unfortunately LoRa Alliance often has bad information.

From the gov agency in charge of regulations in the Philippines (NTC):
- [LORAWAN FREQUENCIES AVAILABLE IN THE PHILIPPINES](https://www.foi.gov.ph/requests/aglzfmVmb2ktcGhyHQsSB0NvbnRlbnQiEE5UQy02OTA2NDUwODE5MzQM) (no mention of 915-918MHz)

Same docs below are referenced in other responses about LoRa:

- [MC-03-06-2017.pdf](https://github.com/meshtastic/firmware/files/12506777/MC-03-06-2017.pdf)
- [MC 03-05-2007 FREQUENCY BANDS FOR THE USE AND OPERATION OF.pdf](https://github.com/meshtastic/firmware/files/12506778/MC.03-05-2007.FREQUENCY.BANDS.FOR.THE.USE.AND.OPERATION.OF.pdf)

There's also an unanswered request specific to AS923-3 availability:
- [Approved LORAWAN Frequency in the Ph following the recent approved Freq by LoRa alliance ](https://www.foi.gov.ph/requests/aglzfmVmb2ktcGhyHQsSB0NvbnRlbnQiEE5UQy0zODQ3ODgxMjgwMjIM)

markbirss, 2023-09-04:
@andrekir 

Ok, thus "03-06-2017" Memorandum applies and confirms the actual LoRa frequency allocation, should we have a PH region setting   or we would simply use EU868 ?

868.000-868.600 (25 mW erp)
868.700-869.200 (25 mW erp)
869.300-869.700 (25 mW erp)
869.700-870.000 (5 mW erp)

[image]
https://github.com/meshtastic/firmware/files/12506777/MC-03-06-2017.pdf

Other findings:
It appears the PH Helium interest group have adopted this same incorrect AS923-3 (915-918) band...

https://www.youtube.com/watch?v=hOBZkBINjlM
https://github.com/helium/miner/blob/master/priv/countries_reg_domains.csv
https://github.com/helium/miner/commit/fb3462e4bd4123074c21b1f5f561209cf522c62d

PH | 12.879721 | 121.774017 | Philippines | AS923_3 | zone1
-- | -- | -- | -- | -- | --

andrekir, 2023-09-04:
yes, PH would need it's own unique region.

CrimsonLustMUFM, 2023-09-06:
Open and unprotected radio frequency bands in the Philippines MC-03-08-2013

[https://region7.ntc.gov.ph/images/LawsRulesAndRegulations/MC/WDN/MC_03-08-2013.pdf](url)

[image]

markbirss, 2023-09-12:
@CrimsonLustMUFM 

Yes, so these regulations in paper were superseded by the later publication - https://github.com/meshtastic/firmware/files/12506777/MC-03-06-2017.pdf

yet the helium roll-out using AS923-3

caveman99, 2023-09-14:
maybe @beegee-tokyo has some insight into this?

beegee-tokyo, 2023-09-14:
All I know is that AS923-3 is the ISM band for Philippines and no-one is paying any fee for devices (beside of the fact that 50PHP and 500PHP are really not much money).
There is one company here that is using EU868 for their private LoRaWAN network. They have a quite good coverage and selling usage of their network to customers.