meshtastic/firmware #2793: [Bug]: Bluetooth pairing always fails

mverch67, 2023-09-13:
### Category

BLE

### Hardware

Other

### Firmware Version

2.2.3 - 2.2.6

### Description

Since firmware 2.2.3 the pairing with Android APP fails for devices when installed using flash erase. The APP constantly tries to re-connect but pairing fails when entering the proper 6 digit pin code.

Workaround:
Install firmware version 2.2.2, then connect with APP via bluetooth. After successful connect an update (but not install with flash erase) to firmware 2.2.3 -2.2.6 is possible.

garthvh, 2023-09-13:
Does forgetting the device also work? Are you checking the checkbox or using the drop down in the web flasher?

mverch67, 2023-09-13:
Forgetting the device does not work. Using CLI device-install.

me639276639936, 2023-09-14:
T-watch, iPhone pairing. Same problem

cordelster, 2023-09-15:
I can confirm, from iPhone app, and Android I am seeing the same issue 2.2.3 to current 2.2.6 on all platforms (techo, heltec, tbeam, tloras).
If anyone needs output from any of these devices, I can add, though it dont look like it specific to device.
I dont have a screen for any of my RAK so not using a random pin there and can pair using default 123456.

liveacid, 2023-09-16:
I am also receiving this issue on the latest Android app release (2.2.4) for firmaware v2.3 on a Heltec LoRa32 v3. I was able to pair one time but was not able to set the region. I have not been able to reproduce a successful pairing since on these virsions. Erasing and installing firmware (v2.2 and below) on the Heltec LoRa32 V3 solved the problem.

closed 2023-09-27 by thebentern

ppyzik, 2024-08-08:
For t3-s3 bug also occurs on firmware versions
2.4.0.46d7b82 Beta
2.3.15.deb7c27 Beta
2.3.13.83f5ba0 Beta
2.3.12.24458a7 Beta
2.4.1.394e0e1 Alpha
2.3.14.64531fa Alpha
2.3.9.f06c56a Alpha

Kabron287, 2026-04-13:
FW 2.7.15(567b8ea) Heltec v3. 2 devices tested.
Bluetooth not active while WIFI is ON.

Is it esp32-s3 "feature" ?
All my ESP32 devices works fine with Bluetooth and WIFI ON.