meshtastic/firmware #2796: [Bug]: WiFI hangs on a nodes with MQTT

Burnwald, 2023-09-14:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

2.2.6.b53cb38

### Description

On all firmwares starting with 2.1.18 while MQTT is used, WiFi connection hangs after some time and requires device reboot. 
Debug log was taken on the last firmware and T-Beam. 
Same issue was observed by many users on T-Beam and DIY devices.
Issue wasn't observed in 2.1.16 and 2.1.17.

### Relevant log output

```Shell
INFO  | 14:50:23 1056 [mqtt] Attempting to connect directly to MQTT server working-mqtt-server.net, port: 1883, username: username, password: 12345678
[1059972][I][WiFiClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 52
ERROR | 14:50:26 1059 [mqtt] Failed to contact MQTT server directly (5/5)...
INFO  | 14:50:26 1060 [WifiConnect] Reconnecting to WiFi acc[1060089][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[1060090][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
ess point ASUS
DEBUG | 14:50:26 1060 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 14:50:26 1060 [WifiConnect] WiFi lost connection. Reason: 8
[1065824][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 14:50:32 1065 WiFi-Event 4: Connected to access point
[1065916][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[1065916][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.18, MASK: 255.255.255.0, GW: 192.168.1.1
DEBUG | 14:50:32 1065 WiFi-Event 7: Obtained IP address: Using non-TLS-encrypted session
INFO  | 14:50:32 1065 Attempting to connect directly to MQTT server working-mqtt-server.net, port: 1883, username: username, password: 12345678
INFO  | 14:50:32 1066 MQTT connected
INFO  | 14:50:32 1066 published online=1
INFO  | 14:50:32 1066 Subscribing to msh/2/c/LongFast/#
INFO  | 14:50:32 1066 [WifiConnect][1066178][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[1066178][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
 Reconnecting to WiFi access point ASUS
DEBUG | 14:50:32 1066 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 14:50:32 1066 [WifiConnect] WiFi lost connection. Reason: 8
```

Fertrius, 2023-09-14:
Same here, still using 2.1.17 with it auto reboots.

ut3uku, 2023-09-14:
I have this problem on TLora 2.1.16 and Meshtastic DIY v1.

infradom, 2023-10-03:
Similar problem here, although I noticed that the MQTT server logs show authentication failed messages while using correct credentials. After a while, Wifi connection is lost

assada, 2023-10-05:
Looks for me like an some issues in MQTT::sendSubscriptions 
Related issue: https://github.com/peterhinch/micropython-mqtt/issues/51#issuecomment-760692967

assada, 2023-10-05:
My Wifi\mqtt only log:

```log
INFO  | 19:02:35 935 [mqtt] Received MQTT topic msh/2/c/LongFast/!25e8b354, len=113
INFO  | 19:02:38 938 [mqtt] Using non-TLS-encrypted session
INFO  | 19:02:38 938 [mqtt] Attempting to connect directly to MQTT server mqtt.secret.server, port: 1883, username: secret_login, password: secret_pass
[941464][I][WiFiClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 52
ERROR | 19:02:41 941 [mqtt] Failed to contact MQTT server directly (1/5)...
INFO  | 19:03:11 971 [mqtt] Using non-TLS-encrypted session
INFO  | 19:03:11 971 [mqtt] Attempting to connect directly to MQTT server mqtt.secret.server, port: 1883, username: secret_login, password: secret_pass
[974494][I][WiFiClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 52
ERROR | 19:03:14 974 [mqtt] Failed to contact MQTT server directly (2/5)...
INFO  | 19:03:44 1004 [mqtt] Using non-TLS-encrypted session
INFO  | 19:03:44 1004 [mqtt] Attempting to connect directly to MQTT server mqtt.secret.server, port: 1883, username: secret_login, password: secret_pass
[1007524][I][WiFiClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 52
ERROR | 19:03:47 1007 [mqtt] Failed to contact MQTT server directly (3/5)...
DEBUG | 19:03:56 1015 [Screen] Screen: Joined: eTaTi
DEBUG | 19:04:03 1022 [GPS] WANT GPS=1
INFO  | 19:04:03 1022 [GPS] Setting GPS power=1
INFO  | 19:04:18 1037 [mqtt] Using non-TLS-encrypted session
INFO  | 19:04:18 1037 [mqtt] Attempting to connect directly to MQTT server mqtt.secret.server, port: 1883, username: secret_login, password: secret_pass
[1040554][I][WiFiClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 52
ERROR | 19:04:21 1040 [mqtt] Failed to contact MQTT server directly (4/5)...
INFO  | 19:04:51 1070 [mqtt] Using non-TLS-encrypted session
INFO  | 19:04:51 1070 [mqtt] Attempting to connect directly to MQTT server mqtt.secret.server, port: 1883, username: secret_login, password: secret_pass
[1073584][I][WiFiClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 52
ERROR | 19:04:54 1073 [mqtt] Failed to contact MQTT server directly (5/5)...
INFO  | 19:04:54 1073 [WifiConnect] Reconnecting to WiFi access point Maxira
[1073733][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[1073736][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 19:04:54 1073 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 19:04:54 1073 [WifiConnect] WiFi lost connection. Reason: 8
[1079513][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 19:04:59 1079 [WebServerThread] WiFi-Event 4: Connected to access point
[1079536][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[1079536][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 10.10.10.63, MASK: 255.255.255.0, GW: 10.10.10.1
DEBUG | 19:05:00 1079 WiFi-Event 7: Obtained IP address: Using non-TLS-encrypted session
INFO  | 19:05:00 1079 Attempting to connect directly to MQTT server mqtt.secret.server, port: 1883, username: secret_login, password: secret_pass
INFO  | 19:05:00 1079 MQTT connected
INFO  | 19:05:00 1079 published 1
INFO  | 19:05:00 1079 Subscribing to msh/2/c/LongFast/#
INFO  | 19:05:00 1079 Subscribing to msh/2/json/LongFast/#
INFO  | 19:05:00 1079 [WifiConnect] Reconnecting to WiFi access point Maxira
[1079883][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[1079886][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 19:05:00 1079 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 19:05:00 1079 [WifiConnect] WiFi lost connection. Reason: 8
[1085659][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 19:05:06 1085 WiFi-Event 4: Connected to access point
[1085692][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[1085692][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 10.10.10.63, MASK: 255.255.255.0, GW: 10.10.10.1
DEBUG | 19:05:06 1085 WiFi-Event 7: Obtained IP address: Using non-TLS-encrypted session
INFO  | 19:05:06 1085 Attempting to connect directly to MQTT server mqtt.secret.server, port: 1883, username: secret_login, password: secret_pass
[1085723][D][WiFiClient.cpp:546] connected(): Disconnected: RES: -1, ERR: 113
INFO  | 19:05:06 1085 [WifiConnect] Reconnecting to WiFi access point Maxira
[1085955][E][WiFiClient.cpp:268] connect(): socket error on fd 52, errno: 113, "Software caused connection abort"
ERROR | 19:05:06 1085 [WifiConnect] Failed to contact MQTT server directly (1/5)...
[1085971][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[1085974][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 19:05:06 1085 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 19:05:06 1086 [WifiConnect] WiFi lost connection. Reason: 8
[1091686][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 19:05:12 1091 WiFi-Event 4: Connected to access point
[1091699][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[1091700][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 10.10.10.63, MASK: 255.255.255.0, GW: 10.10.10.1
DEBUG | 19:05:12 1091 WiFi-Event 7: Obtained IP address: Using non-TLS-encrypted session
[... cut ...]
[1127645][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 10.10.10.63, MASK: 255.255.255.0, GW: 10.10.10.1
INFO  | 19:05:48 1127 [WifiConne[1127925][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[1127926][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
ct] Reconnecting to WiFi access point Maxira
DEBUG | 19:05:48 1127 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 19:05:48 1127 [WifiConnect] WiFi lost connection. Reason: 8
```

assada, 2023-10-06:
Based on my personal experience, the MQTT(pub/sub) and wifi stack on the stm32 operates quite unstably. This is particularly noticeable in large mesh networks where MQTT is a part of the infrastructure supporting crucial network components (bots, statistics). I would propose adding the capability for UART communication and offloading the network operations to devices like RaspberryPI. While this may not be necessary for the average user, it will be very beneficial for supporting mesh infrastructure.

garthvh, 2023-10-08:
> Based on my personal experience, the MQTT(pub/sub) and wifi stack on the stm32 operates quite unstably. This is particularly noticeable in large mesh networks where MQTT is a part of the [...]

We don't have STM32 support working yet but we do already have a MQTT client proxy function where MQTT can be managed by a client instead of the device (this also works on devices without wifi) It has been implemented in the apple apps and the C# CLI, to run it on a raspberry pi the C# CLI should work nicely.

vladbondarenko, 2023-11-05:
same on 2.2.13 on esp32-diyv1

with

[4070765][E][WiFiClient.cpp:422] write(): fail on fd 52, errno: 113, "Software caused connection abort"

Burnwald, 2023-11-09:
Only stable release I stopped at is 2.1.17 because it automatically reboots device and does not require manual reboot.

RoodJuriy77, 2023-11-27:
It looks like the problem reason of variable mqttClientProxyMessagePool. Wifi works stable if clean proxy pool periodically .

stylesuxx, 2023-12-20:
Also having this - or at least a very similar - issue: works fine in Client Router MTTQ mode for a while, then disconnects and never manages to get a connection again:

```
[5887489][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
:31:51 5887 [WifiConnect] Reconnecting to WiFi access point Takbir
DEBUG | 17:31:51 5887 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 17:31:51 5887 [WifiConnect] WiFi lost connection. Reason: 8
[5893341][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 17:31:57 5893 WiFi-Event 4: Connected to access point
INFO  | 17:31:57 5893 [WifiConnect] Reconnecting to WiFi access point Takbir
[5893627][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[5893630][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 17:31:57 5893 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 17:31:57 5893 [WifiConnect] WiFi lost connection. Reason: 8
DEBUG | 17:32:02 5898 [Power] Battery: usbPower=1, isCharging=0, batMv=4183, batPct=100
[5899405][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 17:32:03 5899 [WebServerThread] WiFi-Event 4: Connected to access point
INFO  | 17:32:03 5899 [WifiConnect] Reconnecting to WiFi access point Takbir
[5899694][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[5899697][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 17:32:03 5899 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 17:32:03 5899 [WifiConnect] WiFi lost connection. Reason: 8
[5905501][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 17:32:09 5905 WiFi-Event 4: Connected to access point
INFO  | 17:32:09 5905 [WifiConnect] Reconnecting to WiFi access point Takbir
[5905829][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[5905832][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
DEBUG | 17:32:10 5905 [WifiConnect] WiFi-Event 5: Disconnected from WiFi access point
WARN  | 17:32:10 5905 [WifiConnect] WiFi lost connection. Reason: 8
```

At this point obviously the WebUI stops working, the TBEAM does no longer react to button presses, or if, then very sluggish and randomly. Sometimtes it works fine for a couple hours, sometimes it's only minutes. Reboot fixes the issue.

Even if it's not solving the issues, I think an *optional* watchdog would be nice, allowing to reboot the device after **n** failed attempts of reconnecting, after at least one successful connection.

GUVWAF, 2023-12-20:
I wasn’t able to reproduce a disconnection with MQTT enabled on a T-Beam running 2.2.16. 

However, I believe #3026 fixes the infinite reconnection retries. Can someone verify this with the artifacts from that PR?

mcumings, 2023-12-21:
Did not seem to help my node:

```
INFO  | ??:??:?? 1 Meshtastic hwvendor=4, swver=2.2.17.944e9fc
...
[ 59775][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 03:56:50 59 WiFi-Event 4: Connected to access point
[ 59792][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 59792][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.0.133, MASK: 255.255.255.0, GW: 192.168.0.1
DEBUG | 03:56:51 59 WiFi-Event 7: Obtained IP address: 192.168.0.133
DEBUG | 03:56:56 65 [Power] Battery: usbPower=1, isCharging=1, batMv=3964, batPct=78
[ 67003][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 67004][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 3 - AUTH_LEAVE
[ 67007][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 03:56:58 67 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 03:56:58 67 WiFi lost connection. Reason: 3
INFO  | 03:56:59 68 [WifiConnect] Reconnecting to WiFi access point Scummer
[ 74804][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 74805][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 74808][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 03:57:06 74 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 03:57:06 74 [WebServerThread] WiFi lost connection. Reason: 2
INFO  | 03:57:07 75 [WifiConnect] Reconnecting to WiFi access point Scummer
[ 82536][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 82536][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 82539][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 03:57:13 82 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 03:57:13 82 WiFi lost connection. Reason: 2
INFO  | 03:57:14 83 [WifiConnect] Reconnecting to WiFi access point Scummer
DEBUG | 03:57:19 88 [Power] Battery: usbPower=1, isCharging=1, batMv=3964, batPct=78
[ 90279][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 90280][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 90283][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi AutoReconnect Running
DEBUG | 03:57:21 90 WiFi-Event 5: Disconnected from WiFi access point
WARN  | 03:57:21 90 [WebServerThread] WiFi lost connection. Reason: 2
INFO  | 03:57:22 91 [WifiConnect] Reconnecting to WiFi access point Scummer
[ 97058][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
DEBUG | 03:57:28 97 WiFi-Event 4: Connected to access point
[ 97075][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 97075][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.0.133, MASK: 255.255.255.0, GW: 192.168.0.1
DEBUG | 03:57:28 97 WiFi-Event 7: Obtained IP address: 192.168.0.133
```

GUVWAF, 2023-12-21:
@mcumings Since I wasn't able to reproduce disconnects, I am not sure why this happens. However, you can see that it successfully reconnects now, without requiring a reboot like @stylesuxx and @Burnwald reported.

Burnwald, 2023-12-21:
Dear @GUVWAF 
Thank you for your efforts!
Started testing with latest firmware, and will provide feedback.

stylesuxx, 2023-12-21:
@GUVWAF #3026 seems to have changed something, there are no longer infinite connection retries. WiFi nonetheless stops working at some point. Display shows:

```
WiFi: Not connected
Idle ... Reconnecting
```

Serial log shows no re-connection attempts.

GUVWAF, 2023-12-21:
@stylesuxx Can you show the serial logs? It should still try to reconnect, although at a lower rate.

stylesuxx, 2023-12-21:
@GUVWAF you are right, seems I was too inpatient, it just reconnected. But there still seems to be something wonky: It reconnects, I can see a new DHCP lease on the router, but I can't access the device via web UI and the device also shows DNS errors in the serial log when trying to resolve the address for the MQTT server. I just started a new run and will post a complete log once I arrive at this deadlock again.

PR #3026 (2023-12-21): ESP32: Don't disconnect WiFi when we're actively reconnecting
The WiFiEvent `ARDUINO_EVENT_WIFI_STA_DISCONNECTED` or `ARDUINO_EVENT_WIFI_STA_LOST_IP` may happen when we're about to reconnect. It shouldn't disconnect and schedule new reconnects then.

Might partly fix #2796.
closed 2023-12-21 by thebentern