meshtastic/firmware #2797: [Bug]: Despite disabled air quality module, the data is displayed

memo-567, 2023-09-15:
### Category

Hardware Compatibility

### Hardware

T-Beam

### Firmware Version

2.2.6.b53cb38 Alpha

### Description

Despite disabled air quality module, the data is displayed both, locally and via mesh.

Sensor: BME680

[image] 

[image]

garthvh, 2023-09-15:
That data is part of the environment metrics if you have a 680. Air quality is a different module that uses AQI specific features.

memo-567, 2023-09-15:
Ah okay. Thanks for the info 😀

closed 2023-09-15 by memo-567