meshtastic/firmware #2801: [Bug]: GPS is never detected

D4rk4, 2023-09-16:
### Category

Hardware Compatibility, Other

### Hardware

Rak4631

### Firmware Version

2.2.6.b53cb38

### Description

Look like regression.
GPS module is EVA-M8E-0.

jp-bennett, 2023-09-16:
Anything interesting in the log output? Does the device eventually start getting GPS updates? We try to fail safe by listening at 9600 baud even with nothing found. Is that GPS available on a Rak module?

D4rk4, 2023-09-16:
It appears that only the monteops_hw1 variant is malfunctioning. I have conducted tests using various firmware versions that incorporate the new GPS detection logic and discovered that the native rak4631 firmware variant operates correctly.

jp-bennett, 2023-09-17:
Are any of these available as hardware samples?

D4rk4, 2023-09-18:
@jp-bennett I will send you one of them when the next test batch arrives.

closed 2024-03-28 by thebentern