meshtastic/firmware #2815: [Bug]: T-echo GPS data not updating

swiftarf, 2023-09-20:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.1.23.04bbdc6

### Description

T-echo in tracker mode, set to send position every 10s for testing. Router receives LoRa messages from tracker at 10s intervals but exactly the same GPS location sent multiple times (at least 2 or 3) before correct location sent. GPS has good view of sky, power saving turned off or to long time, GPS updates set to 1sec. I suspect GPS is entering sleep mode after receiving a location despite update setting.

jp-bennett, 2023-09-26:
There aren't any planned future releases in the 2.1.x line, and we're in the midst of a GPS revamp, which will likely address this issue.

swiftarf, 2023-09-26:
Ok, thanks for that!

closed 2023-09-26 by swiftarf