meshtastic/firmware #2816: [Bug]: 2.2.7 pico does not initialize properly

NickDolfin, 2023-09-20:
### Category

Other

### Hardware

Raspberry Pi Pico (W)

### Firmware Version

2.2.7.e8970ad

### Description

It hangs during boot. 
The led stays on continuesly.
I cannot get a putty window on the serial port.

The previous alpha version 2.2.6 works

thebentern, 2023-09-20:
GPS initialization issue possibly based on that last line. @jp-bennett do you have a pico based board?

jp-bennett, 2023-09-20:
No, but I'd agree, it's something odd with trying to power up GPS. Which may or may not exist. What is the general setup for a pico board? Is there a Lora HaT?

thebentern, 2023-09-20:
There's also the RAK RP2040 based core but don't know if it's affected.

jp-bennett, 2023-09-20:
We dropped the check for GPS_SERIAL_NUM during init. That chip probably doesn't have a serial port to try to bring up. I'll open a PR just to try to fix this in a bit.

NickDolfin, 2023-09-20:
I have a Waveshare RP2040-Plus 4MB coupled to a Waveshare SX1262 LoRa Node Module voor Raspberry Pi Pico.

No gps at the moment. Perhaps in the future.

jp-bennett, 2023-09-20:
In theory #2817 fixes this hang.

NickDolfin, 2023-09-21:
Nope, 2.2.8 is not working.

I tried to capture the output:
▒▒@INFO  | ??:??:?? 1

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 1 Filesystem files:
DEBUG | ??:??:?? 1  channels.proto (53 Bytes)
lfs_file_close: fd=0x20012e14
                             DEBUG | ??:??:?? 1  config.proto (93 Bytes)
lfs_file_close: fd=0x20012e14
                             DEBUG | ??:??:?? 1  db.proto (299 Bytes)
lfs_file_close: fd=0x20012e14
                             DEBUG | ??:??:?? 1  module.proto (88 Bytes)
lfs_file_close: fd=0x20012e14
                             DEBUG | ??:??:?? 1 Using analog input 26 for batter                                                                                                               y level
INFO  | ??:??:?? 1 Scanning for i2c devices...
DEBUG | ??:??:?? 1 Scanning for i2c devices on port 1
INFO  | ??:??:?? 1 No I2C devices found
DEBUG | ??:??:?? 1 acc_info = 0
INFO  | ??:??:?? 1 Meshtastic hwvendor=47, swver=2.2.8.8f9d335
INFO  | ??:??:?? 1 Initializing NodeDB
INFO  | ??:??:?? 1 Loading /prefs/db.proto
lfs_file_close: fd=0x20012bb4
                             INFO  | ??:??:?? 1 Loaded saved devicestate version                                                                                                                22
INFO  | ??:??:?? 1 Loading /prefs/config.proto
lfs_file_close: fd=0x20012bb4
                             INFO  | ??:??:?? 1 Loaded saved config version 22
INFO  | ??:??:?? 1 Loading /prefs/module.proto
lfs_file_close: fd=0x20012bb4
                             INFO  | ??:??:?? 1 Loaded saved moduleConfig versio                                                                                                               n 22
INFO  | ??:??:?? 1 Loading /prefs/channels.proto
lfs_file_close: fd=0x20012bb4
                             INFO  | ??:??:?? 1 Loaded saved channelFile version                                                                                                                22
LittleFSDirImpl::openFile: rc=-2 fd=0x20012bb4 path=`/oem/oem.proto` openMode=0                                                                                                                accessMode=1 err=-2
                   INFO  | ??:??:?? 1 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 1 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 1 Expanding short PSK #1
INFO  | ??:??:?? 1 Wanted region 3, using EU_868
DEBUG | ??:??:?? 1 region=3, NODENUM=0xdf58312c, dbsize=3
DEBUG | ??:??:?? 1 Read RTC time as 0
DEBUG | ??:??:?? 1 WANT GPS=1

Green led is still on.

NickDolfin, 2023-09-21:
@jp-bennett Yes, you are right. I was too fast and downloaded the wrong one.

Now 2.2.8 is working.

Many thanks for the fast fix.

closed 2023-09-21 by jp-bennett