meshtastic/firmware #2818: [Bug]: tWatch s3 screen never sleeps with newest prerelease

me639276639936, 2023-09-21:
### Category

Hardware Compatibility, Other

### Hardware

T-Watch

### Firmware Version

2.2.7.e8970ad 

### Description

After updating to latest prelease, watch screen does not sleep. Always on.

mverch67, 2023-09-21:
Can you please check the display settings. What value is set for "Screen timeout" ?

me639276639936, 2023-09-21:
I see. I guess that is something I have to update after the update. 

Thanks!! Works now and I feel dumb. Hahaha

closed 2023-09-21 by thebentern

mverch67, 2023-09-21:
> I see. I guess that is something I have to update after the update.
> 
> Thanks!! Works now and I feel dumb. Hahaha

You probably did a flash erase during the update witch sets all values to default (= no screen timeout).