meshtastic/firmware #2822: [Feature Request]: Implement deep sleep for NRF52

andrekir, 2023-09-23:
### Platform

NRF52

### Description

RAK suggests:

### `xSemaphoreTakeBinary(semphoreHandle, portMAX_DELAY)`

FreeRTOS provides `semaphores` to control task switches and let tasks "sleep" while waiting for an event. Looking into the [FreeRTOS documentation](https://www.freertos.org/a00113.html), you can see there are several types of semaphores, named `binary`, `counting`, `mutex` and `recursive mutex`. To keep things simple here, I will use the `binary` semaphore in this example.

source: https://github.com/RAKWireless/WisBlock/blob/master/examples/RAK4630/communications/LoRa/LoRaWAN/Low_Power_Example.md

---

if this is something that could be implemented it would allow, for example, SENSORS running for very long periods (months?) without a charge.

jp-bennett, 2023-09-24:
If I understand it, this would mean converting our entire codebase away from using the Arduino threads concurrency library, to using full RTOS, which is very involved.

andrekir, 2023-09-25:
got it, thanks. so the FreeRTOS method is not viable.

right now we're only using low-level `NRF_POWER->SYSTEMOFF = 1;` for deep sleep and [it's not using the `msecToWake` parameter](https://github.com/meshtastic/firmware/blob/master/src/platform/nrf52/main-nrf52.cpp#L158-L198) (ie. to wake up after N seconds).

the other examples I found require an RTC to generate an interrupt and not sure our RTC lib supports that.

jp-bennett, 2023-09-25:
Would we do nearly as well, to just power everything down, like the lora chip, GPS, BLE, and LEDs, and then do a long delay() call?

andrekir, 2023-09-25:
that's a great idea. perhaps we can get into sub-mA power draw using these [NRF power states](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v3.0.0%2Fgroup___n_r_f___s_o_c___e_n_u_m_s.html&anchor=ga66c4a4f3902bf8dfd3751c32e6589790):

```
sd_power_mode_set(NRF_POWER_MODE_LOWPWR) // low power
delay(secToWake)
sd_power_mode_set(NRF_POWER_MODE_CONSTLAT) // default
```

also, this might need to go into NRF52 Light Sleep logic (for TRACKERs and SENSORs).

andrekir, 2023-09-28:
for simplicity sake, going for RTC awake on this request, then open a separate one for low-power states for TRACKER/SENSOR roles.

current RTC libs: [PCF8563](https://github.com/lewisxhe/PCF8563_Library) & [Melopero_RV-3028](https://github.com/melopero/Melopero_RV-3028_Arduino_Library).

PCF8563 example: https://github.com/Xinyuan-LilyGO/T-Echo/blob/main/examples/Sleep/Sleep.ino

closed 2024-10-07 by thebentern