meshtastic/firmware #2825: [Bug]: Pico rp2040 serial output not working

NickDolfin, 2023-09-24:
### Category

Serial

### Hardware

Raspberry Pi Pico (W)

### Firmware Version

2.2.8

### Description

Dear programmer,

I am trying to get a serial output port to work.
I tried pins gp4+gp5 (uart1). There is some activity on reboot, but looks like gibberisch on a (serial) terminal like putty or 'feasycom serial port' program
In addition, when I use different pins eg. go0/gp1 (uart0) or gp8/gp9 (uart1) no output is seen.

I used e.g:

meshtastic --set serial.enabled true --set serial.echo true --set serial.mode TEXTMSG --set serial.baud BAUD_115200 --set serial.rxd 9 --set serial.txd 8
Connected to radio
Set serial.enabled to true
Set serial.echo to true
Set serial.mode to TEXTMSG
Set serial.baud to BAUD_115200
Set serial.rxd to 9
Set serial.txd to 8
Writing modified preferences to device

Question? Is the serial output available in the rp2040?
If yes, how do I get it to work?

Thanks

GUVWAF, 2023-09-24:
Unfortunately the SerialModule is not yet available on the RP2040.

NickDolfin, 2023-09-24:
Thanks,

I just wasted a whole afternoon. These info should be in the manual....

mverch67, 2023-09-25:
If you disable bluetooth for the T3S3 (which you probably don't have enabled at the rp2040) and switch off the display and set the CPU frequency to 80MHz then you're also in the range of ~38mA for the T3S3. And when entering light sleep (with LoRa module enabled) you are below 15mA.

GUVWAF, 2023-09-25:
Yes, and nRF52-based boards can go even lower than that.
That aside, I can look into getting the SerialModule to work on the Pico.

GUVWAF, 2023-09-25:
@NickDolfin It works for me using PR #2830 with `serial.txd 8` and `serial.rxd 9`.

NickDolfin, 2023-09-25:
@GUVWAF Wow, thats nice.

Any idea where /how I can download the uf2 file for me to test it?

GUVWAF, 2023-09-25:
Yes, you can find it here for the Pico or Pico W: https://github.com/meshtastic/firmware/actions/runs/6302933560#artifacts

NickDolfin, 2023-09-25:
Many thanks.

Finaly I have something working in the rp2040 using serial.txd 8 and serial.rxd 9.

Few observations:

I attached a ftdi cable on those pins. On my scope I saw that the polarity of the signals was wrong. I used the FT_Prog to change the polarity of both the txd and rxd levels. Now the polarity is idle high for both signals. (see jpg).
I can receive and send using mode 'Textmsg'.
There are some additional bytes send before the real message. I'll have to look into that. Is this a bug?

The 2.2.9 firmware is skipping data. Not all characters are received.
When I send (through an 2.2.7 esp32 attached to Ipad) the text "The quick brown lazy fox jumps over the river" everything is displayed on my rp2040 terminal and also decoded by my scope. But when sending from the rp2040, on the esp the following is received: " The quick brown lazy fox jumps o". See jpg of the terminal and scope

[image]
[image]
[image]
[image]

NickDolfin, 2023-09-25:
Any idea?

closed 2023-09-26 by thebentern

NickDolfin, 2023-09-26:
I still do not know why the serial signals are inverted.
Is this correct? It this standard?

I would like to use the serial signals to attach to a bluetooth to serial uart device like a HC-02.
Is there an option to change the mode of the levels? I think my ftdi cable is in its standard setting the default for many chips including the aforementioned hc-05 or hc02.

GUVWAF, 2023-09-26:
It’s using the standard HardwareSerial provided by the Pico core (https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/SerialUART.h#L34), which has no method to invert the polarity. It is correctly interfacing with my ESP8266 like this.

The missing characters is indeed caused by another change in 2.2.8 as you already found out.

NickDolfin, 2023-09-26:
Thanks for the clarification.
I just looked up the uart in the rp2040 datasheet and it also tells that idle is high.
Then I do not understand why the ftdi cable used as standard uses the wrong polarity. I had to explicitly invert them using the programmer from ftdi. Perhaps I am making somewhere a mistake?

PR #2830 (2023-09-26): RP2040: Add SerialModule support 
Fixes #2825. Tested with `serial.txd 8` and `serial.rxd 9`. 

Had to increase the `RX_BUFFER` to 256, otherwise long packets would occasionally fail. I saw that for ESP32 it even [throws an error]( https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.cpp#L580.) if you have it set to 128 or lower.