meshtastic/firmware #2826: [Bug]: Locations sent from phone api are getting garbled

garthvh, 2023-09-24:
### Category

Other

### Hardware

T-Lora v2 1.6, T-Echo, Rak4631

### Firmware Version

2.2.7

### Description

I have verified that my send location to device functionality is pushing in appropriate position messages to the device but I am getting repeated output of the same coordinates and timestamp.

garthvh, 2023-09-24:
Sent to Device 

Meshtastic.Position:
latitude_i: XXXX
longitude_i: -XXX
altitude: 97
time: 1695588481
timestamp: 1695588481
sats_in_view: 12

Send from device to mesh 

Meshtastic.Position:
latitude_i: 476022033
longitude_i: -122
altitude: 97
time: 1695573920

thebentern, 2023-10-10:
Should be fixed in 2.2.11. Will reopen if it occurs again

closed 2023-10-10 by thebentern