meshtastic/firmware #2829: [Bug]: T-Lora Lilygo T3S3 2.2.8 reboot not working

NickDolfin, 2023-09-25:
### Category

Other

### Hardware

Other

### Firmware Version

2.2.8

### Description

After changing parameters in using firmware 2.2.7 the device reboots automatically. Good!
This does not happen in 2.2.8. I have to use the reset button. In addition changing to a power supply that is not attached to a computer results in no startup.

I have no idea what to do. I reflashed the older 2.2.7 in it and it works flawlessly.

thebentern, 2023-12-09:
Is this issue still occurring?

NickDolfin, 2023-12-09:
* I do not know. I have not checked it lately.*

sergeybe2, 2023-12-15:
This bug is actual for 2.2.14.

thebentern, 2023-12-16:
Just tested rebooting via the admin command and implicit reboot on config change for my T3S3. It's working flawlessly. Something is likely incorrect with your configuration or setup.

closed 2023-12-16 by thebentern