meshtastic/firmware #2832: [Bug]: GPS Module not found on Nano G1 Explorer

Eric-Sparks, 2023-09-26:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.2.7.e8970ad.alpha

### Description

When upgrading from 2.2.4 alpha, and before, to 2.2.7 alpha the GPS module on the Nano G1 Explorer isn't detected.  I've been able to recreate this problem on two of four devices I have.  Upgrading to the latest firmware (2.2.8 alpha) does not fix the problem.

jp-bennett, 2023-09-26:
Can you check the position configuration, whether GPS_RX_PIN and GPS_TX_PIN are set differently in the units that are failing to detect GPS? The boot log output of the working and non-working devices would be helpful as well. Thanks!

jp-bennett, 2023-10-04:
I believe this should be fixed in the 2.2.10 Alpha release. Let us know if there are further problems.

closed 2023-10-04 by jp-bennett