meshtastic/firmware #2833: [Bug]: Messages sent from 2.2.8 alpha to a lower version are abbreviated

Eric-Sparks, 2023-09-26:
### Category

Other

### Hardware

Other

### Firmware Version

2.2.8.61f6fb2.alpha

### Description

Messages sent from 2.2.8 alpha to another device running a previous version are cut short.  The full message appears on the device's screen but not all the text is sent over the serial connection (BLE or USB).

This is on the Nano G1 Explorer.

### Relevant log output

```Shell
from: 527445864
to: 527445896
decoded {
  portnum: TEXT_MESSAGE_APP
  payload: "THEQUICKBROWNFOXJUMPSOVERTH"
}
id: 198257801
rx_time: 1695698186
rx_snr: 6.0
hop_limit: 3
want_ack: true
rx_rssi: -9
```

Eric-Sparks, 2023-09-26:
Oh, the whole message sent was "THEQUICKBROWNFOXJUMPSOVERTHELAZYDOG".  It also happens on much shorter messages as well but there doesn't seem to be a specific character limit.  Some messages get cut off while really short messages don't.

jp-bennett, 2023-09-26:
@GUVWAF related to compression, maybe?

GUVWAF, 2023-09-26:
Oh, yeah, didn’t realize this is a breaking change… 

This is because it is really the first time compression is actually used. While decompressing, the length wasn’t changed ([this line](https://github.com/meshtastic/firmware/blob/master/src/mesh/Router.cpp#L362) was missing), so devices using <2.2.8 will use the compressed length, which is shorter.

I’m afraid we have to undo #2819 (or remove compression as it wasn’t used at all) until release 3.0.

NickDolfin, 2023-09-26:
I also noticed this: https://github.com/meshtastic/firmware/issues/2825

kokroo, 2023-09-26:
@GUVWAF Shouldn't the decompression happen only for portnum 7? This shouldn't be a breaking change, since all previous versions will stick to only portnum 1. Compressed messages will only go over portnum 7 and will only be decompressed if something is received over portnum 7.

Am I missing something here?

GUVWAF, 2023-09-27:
No, v2.2 is backwards compatible. Besides, 2.2.0 is already released, so keeping the feature means older and never versions of 2.2.x will not be compatible.

closed 2023-10-03 by thebentern