meshtastic/firmware #2838: [Board]: Lilygo T-BEAM 1.2

stefa168, 2023-09-27:
### SOC

ESP32

### Lora IC

LILYGO LORA32 

### Product Link

https://www.amazon.it/dp/B0B454TB6S

### Description

Hi!
Today I received a LILYGO T-BEAM-V1.1 from Amazon, however after further inspection and research because of the GPS not working (tested also with .ino sketches and other stuff which returned no-sense data) I noticed that the power management IC has been changed from an AXP192 to an AXP2101: https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_TBeam_V1.2.pdf

The current firmware (`firmware-tbeam-2.1.23.04bbdc6.bin`) works fine, however the GPS module doesn't work, neither its LEDs blink at all.

Could this be linked to the change of the AXP192?

InvisibleSymbol, 2023-09-27:
I also have a recently bought a T-Beam 1.2 that has non-functioning GPS, although I don't think its related to the PMIC as the code and serial logs do seem to indicate correctly detection and handling as a AXP2101.

here are logs from the GPS section of boot, which shows the bogus values parsed from the GPS module. they are pretty much random each reset.
```
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600 
INFO  | ??:??:?? 2 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 2 [GPS] Module Info : 
DEBUG | ??:??:?? 2 [GPS] Soft version: 7.03 (45969)
DEBUG | ??:??:?? 2 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 2 [GPS] Extensions:0
WARN  | ??:??:?? 3 [GPS] Unable to save GNSS module configuration.
INFO  | ??:??:?? 3 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 3 [GPS] NMEA GPS time 2060-50-104 23:24:03
DEBUG | ??:??:?? 3 [GPS] Upgrading time to quality 4
DEBUG | 23:24:03 3 [GPS] Read RTC time as -1457396864
--- reset ---
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600 
INFO  | ??:??:?? 2 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 2 [GPS] Module Info : 
DEBUG | ??:??:?? 2 [GPS] Soft version: 7.03 (45969)
DEBUG | ??:??:?? 2 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 2 [GPS] Extensions:0
WARN  | ??:??:?? 3 [GPS] Unable to save GNSS module configuration.
INFO  | ??:??:?? 3 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 3 [GPS] NMEA GPS time 2036-80-107 179:75:97
--- reset ---
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600 
INFO  | ??:??:?? 2 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 2 [GPS] Module Info : 
DEBUG | ??:??:?? 2 [GPS] Soft version: 7.03 (45969)
DEBUG | ??:??:?? 2 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 2 [GPS] Extensions:0
WARN  | ??:??:?? 3 [GPS] Unable to save GNSS module configuration.
INFO  | ??:??:?? 3 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 4 [GPS] NMEA GPS time 2004-64-212 245:68:07
```

jp-bennett, 2023-09-28:
Unfortunately I don't have one of these boards to test, but I do have a theory and patchset to test: https://github.com/meshtastic/firmware/pull/2843

The theory is that our code enabled the ram backup line first, which potentially tells the GPS chip to do a warm boot, which we really don't actually want on first boot. If you get a chance, flash the firmware artifact in that PR, then turn the T-Beam off and yank the battery for 10 minutes or so. That should fully drain the power out of the GPS, and ensure a true cold boot, hopefully giving us a sane state.

stefa168, 2023-09-28:
Hi @jp-bennett , thanks for the fast answer and the potential fix.
I flashed it to my board and followed your instructions (waited for an hour), however the LED next to the ublox GPS module still doesn't turn on, while the other two leds under the OLED screen blink at the same time.
The log returns this:

```log
DEBUG | ??:??:?? 1 [GPS] Probing for GPS at 9600 
INFO  | ??:??:?? 2 [GPS] Found a UBlox Module using baudrate 9600
DEBUG | ??:??:?? 2 [GPS] Module Info : 
DEBUG | ??:??:?? 2 [GPS] Soft version: 7.03 (45969)
DEBUG | ??:??:?? 2 [GPS] Hard version: 00040007
DEBUG | ??:??:?? 2 [GPS] Extensions:0
WARN  | ??:??:?? 3 [GPS] Unable to save GNSS module configuration.
INFO  | ??:??:?? 3 [RangeTestModule] Range Test Module - Disabled
DEBUG | ??:??:?? 4 [GPS] NMEA GPS time 2023-09-28 11:12:35
DEBUG | ??:??:?? 4 [GPS] Upgrading time to quality 4
DEBUG | 11:12:35 4 [GPS] Read RTC time as 1695899555
DEBUG | 11:12:35 4 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 11:12:35 4 [GPS] No GPS lock
DEBUG | 11:12:35 4 [GPS] onGPSChanged() pos@0, time=1695899555, lat=0, lon=0, alt=0
INFO  | 11:12:35 4 [GPS] updatePosition LOCAL pos@0, time=1695899555, latI=0, lonI=0, alt=0
DEBUG | 11:12:35 4 [GPS] Node status update: 1 online, 1 total
```
It indeed looks like the ESP32 manages to communicate with the GPS module, however the data makes no sense, especially the part regarding the module info.
I'm attaching the startup log if it can be of any use.

~ Stefano

[meshtastic_1.log](https://github.com/meshtastic/firmware/files/12748759/meshtastic_1.log)

jp-bennett, 2023-09-28:
> It indeed looks like the ESP32 manages to communicate with the GPS module, however the data makes no sense, especially the part regarding the module info.\nI'm attaching the startup log if it can be of any use.

Thanks for trying it! That actually looks really normal. That's what module info should show for this device. And most importantly, GPS is showing the right time. If you give this long enough, it will probably get a lock. The little tiny GPS antenna that comes stock on these is known to not be great, too. Still looking into this.

stefa168, 2023-09-28:
Gotcha! I'll get it near the window

jp-bennett, 2023-09-28:
Second debug build is available at #2843

stefa168, 2023-09-29:
Yesterday I installed the second build and when I came back home I left the T-BEAM near a window (always after waiting for the device to drain its power)
It received GPS signal and managed to return a correct position!

jp-bennett, 2023-10-04:
The fixes should be fully present in the 2.2.10 Alpha release, so closing for now. Let us know if you see further problems.

closed 2023-10-04 by jp-bennett

stefa168, 2023-10-04:
I've been testing the radio for the last days, and it looks like it works fine.
The only thing I noticed is that it takes 15-20 minutes after a cold boot to lock onto the satellites. Is it normal?

kami4ka, 2023-10-15:
Can't make GPS work on 2.2.11 with TBeam 1.2
Flashed via Web Flasher using V2.2.11.10265aa alpha
No sats message always
The screen can't also display non-english characters

jp-bennett, 2023-10-15:
> Can't make GPS work on 2.2.11 with TBeam 1.2

Is there any way you can get a bootup log output using meshtastic --noproto? These devices are particularly terrible at getting the first GPS lock, but seem to work rather well after that. Along with the really weak factory GPS antenna, it's quite a pain to get it going. Sitting it outside with a good view of the sky can help.

kami4ka, 2023-10-16:
@jp-bennett yes, I tried to go outside and test it there as well, as it has worked near the window with different firmware (stock one, 2.2.10, 2.2.11) for the last week - still no luck. I also tried to flash it using Platformio from the master branch, but can't flash the board. It's off-topic, but I've seen some reports that the board should be fully charged (internal batteries and li-ion) to flash it.

I'm not sure whether it's the right log, but there is a piece I was able to catch:

```
DEBUG | ??:??:?? 1 [Screen] Screen: Started...
DEBUG | ??:??:?? 1 [GPS] NMEA GPS time 2023-10-16 06:56:56
DEBUG | ??:??:?? 1 [GPS] Upgrading time to quality 4
DEBUG | 06:56:56 1 [GPS] Read RTC time as 1697439416
WARN  | 06:56:56 1 [GPS] Warning, 1 new GPS checksum failures, for a total of 1.
DEBUG | 06:56:56 1 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 06:56:56 1 [GPS] No GPS lock
DEBUG | 06:56:56 1 [GPS] onGPSChanged() pos@0, time=1697439416, lat=0, lon=0, alt=0
INFO  | 06:56:56 1 [GPS] updatePosition LOCAL pos@0, time=1697439416, latI=0, lonI=0, alt=0
DEBUG | 06:56:56 1 [GPS] Node status update: 1 online, 2 total
INFO  | 06:56:56 1 [RangeTestModule] Range Test Module - Disabled
DEBUG | 06:56:58 3 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=44, time 755 ms
DEBUG | 06:56:58 3 [RadioIf] Lora RX (id=0x3a56df1e fr=0x14 to=0xff, WantAck=0, HopLim=0 Ch=0x62 encrypted rxSNR=12.25 rxRSSI=-64)
DEBUG | 06:56:58 3 [RadioIf] AirTime - Packet received : 755ms
DEBUG | 06:56:58 3 [Router] Add packet record (id=0x3a56df1e fr=0x14 to=0xff, WantAck=0, HopLim=0 Ch=0x62 encrypted rxSNR=12.25 rxRSSI=-64)
WARN  | 06:56:58 3 [Router] No suitable channel found for decoding, hash was 0x62!
DEBUG | 06:56:58 3 [Router] packet decoding failed or skipped (no PSK?) (id=0x3a56df1e fr=0x14 to=0xff, WantAck=0, HopLim=0 Ch=0x62 encrypted rxtime=1697439418 rxSNR=12.25 rxRSSI=-64)
DEBUG | 06:56:58 3 [Router] No modules interested in portnum=-116457444, src=REMOTE
INFO  | 06:56:59 4 [PowerFSM] Initialise the NimBLE bluetooth module
INFO  | 06:57:01 7 [Screen] Done with boot screen...
DEBUG | 06:57:01 7 [Screen] showing standard frames
DEBUG | 06:57:01 7 [Screen] Showing 0 module frames
DEBUG | 06:57:01 7 [Screen] Total frame count: 103
DEBUG | 06:57:01 7 [Screen] Added modules.  numframes: 0
DEBUG | 06:57:01 7 [Screen] Finished building frames. numframes: 3
DEBUG | 06:57:02 7 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=53, time 870 ms
DEBUG | 06:57:02 7 [RadioIf] Lora RX (id=0x6e31a640 fr=0x90 to=0xff, WantAck=0, HopLim=3 Ch=0x62 encrypted rxSNR=2.5 rxRSSI=-99)
DEBUG | 06:57:02 7 [RadioIf] AirTime - Packet received : 870ms
DEBUG | 06:57:02 7 [Router] Add packet record (id=0x6e31a640 fr=0x90 to=0xff, WantAck=0, HopLim=3 Ch=0x62 encrypted rxSNR=2.5 rxRSSI=-99)
WARN  | 06:57:02 7 [Router] No suitable channel found for decoding, hash was 0x62!
DEBUG | 06:57:02 7 [Router] packet decoding failed or skipped (no PSK?) (id=0x6e31a640 fr=0x90 to=0xff, WantAck=0, HopLim=3 Ch=0x62 encrypted rxtime=1697439422 rxSNR=2.5 rxRSSI=-99)
DEBUG | 06:57:02 7 [Router] No modules interested in portnum=635109413, src=REMOTE
DEBUG | 06:57:04 9 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=53, time 870 ms
DEBUG | 06:57:04 9 [RadioIf] Lora RX (id=0x6e31a640 fr=0x90 to=0xff, WantAck=0, HopLim=2 Ch=0x62 encrypted rxSNR=12.75 rxRSSI=-62)
DEBUG | 06:57:04 9 [RadioIf] AirTime - Packet received : 870ms
DEBUG | 06:57:04 9 [Router] Found existing packet record for fr=0x860cb90,to=0xffffffff,id=0x6e31a640
DEBUG | 06:57:04 9 [Router] Found existing packet record for fr=0x860cb90,to=0xffffffff,id=0x6e31a640
DEBUG | 06:57:04 9 [Router] Add packet record (id=0x6e31a640 fr=0x90 to=0xff, WantAck=0, HopLim=2 Ch=0x62 encrypted rxSNR=12.75 rxRSSI=-62)
DEBUG | 06:57:04 9 [Router] Ignoring incoming msg, because we've already seen it (id=0x6e31a640 fr=0x90 to=0xff, WantAck=0, HopLim=2 Ch=0x62 encrypted rxSNR=12.75 rxRSSI=-62)
```

Once flashed, I've seen the message that GPS module required factory reset.

kami4ka, 2023-10-17:
UPD: so GPS started working after several days

dizcza, 2023-11-22:
Is GPS PPS pin available for the user on this board?

p3ntaxian, 2024-01-04:
Hello, I have Lilygo T-BEAM 1.2 with firmware-tbeam-2.2.10.7cebd79 but GPS still does not work. No sats, just sometime flashing red light next to GPS module.

DEBUG | 11:18:14 891 [GPS] WANT GPS=1
WARN  | 11:18:14 891 [GPS] SOME data is TOO OLD: LOC 421205, TIME 421205, DATE 421212
DEBUG | 11:18:14 891 [GPS] WANT GPS=0
DEBUG | 11:18:14 891 [GPS] GPS Lock took 0, average 0
DEBUG | 11:18:14 891 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 11:18:14 891 [GPS] onGPSChanged() pos@0, time=1704367094, lat=0, lon=0, alt=0
INFO  | 11:18:14 891 [GPS] updatePosition LOCAL pos@0, time=1704367094, latI=0, lonI=0, alt=0
DEBUG | 11:18:14 891 [GPS] Node status update: 2 online, 2 total
DEBUG | 11:18:19 896 [GPS] WANT GPS=1
WARN  | 11:18:19 896 [GPS] SOME data is TOO OLD: LOC 426332, TIME 426332, DATE 426339
DEBUG | 11:18:19 896 [GPS] WANT GPS=0
DEBUG | 11:18:19 896 [GPS] GPS Lock took 0, average 0
DEBUG | 11:18:19 896 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 11:18:19 896 [GPS] onGPSChanged() pos@0, time=1704367099, lat=0, lon=0, alt=0
INFO  | 11:18:19 896 [GPS] updatePosition LOCAL pos@0, time=1704367099, latI=0, lonI=0, alt=0
DEBUG | 11:18:19 896 [GPS] Node status update: 2 online, 2 total
DEBUG | 11:18:24 901 [GPS] WANT GPS=1
WARN  | 11:18:24 901 [GPS] SOME data is TOO OLD: LOC 431453, TIME 431453, DATE 431460
DEBUG | 11:18:24 901 [GPS] WANT GPS=0
DEBUG | 11:18:24 901 [GPS] GPS Lock took 0, average 0
DEBUG | 11:18:24 901 [GPS] publishing pos@0:2, hasVal=0, Sats=0, GPSlock=1
DEBUG | 11:18:24 901 [GPS] onGPSChanged() pos@0, time=1704367104, lat=0, lon=0, alt=0
INFO  | 11:18:24 901 [GPS] updatePosition LOCAL pos@0, time=1704367104, latI=0, lonI=0, alt=0
DEBUG | 11:18:24 901 [GPS] Node status update: 2 online, 2 total

I have tried also latest beta or alfa firmware, but still GPS not working.

garthvh, 2024-01-04:
Looks to have found the hardware, are you inside?

p3ntaxian, 2024-01-04:
Lilygo T-BEAM 1.2 is located outside in the garden with visibility to full hemisphere for several hours. No sats.

p3ntaxian, 2024-01-04:
Finally I have found where is problem. Problem is in IOS APP  Meshtastic. If you keep default Position Config in IOS APP, GPS will work and every 15 minutes will send update. However if you change anything in IOS APP for Position Config, GPS will stop working. IOS APP will set "GPS Update Interval" and "Fix Attempt Duration" to 0.  I have verified it with client.meshtastic.org web config tool. So any change for Position Config have to be done by web config tool, not by IOS App. It would be nice to inform about this issue developers of IOS App, what is best way to inform them?

jp-bennett, 2024-01-04:
@p3ntaxian This is possibly related to [protobufs/#407](https://github.com/meshtastic/protobufs/issues/407).