meshtastic/firmware #2844: [Bug]:

NickDolfin, 2023-09-28:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.9.47301a5 alpha

### Description

The red gps led in my Tbeam (with m8n chip) is not blinking anymore using the 2.2.9 alpha firmware.
Using the older firmwares the led does blink properly.

jp-bennett, 2023-09-28:
This is probably expected, with the new GPS powersave code. Do you have GPS position enabled on your T-beam, and is it working as expected?

NickDolfin, 2023-10-02:
Hi,

I just tested it briefly.

Apparantly, the led flashes after a position has been detected wile using the gps update interval.
I think it works as it should.

I will do more testing to fully analyze the gps update interval...

Stay tuned for an update.

NickDolfin, 2023-10-02:
small update:

It appeasr that the gps update rate is correct.
However, the reported (with an ipad attached to a different esp32 board) number of sats is not correct: It remeains at 0 (zero).

NickDolfin, 2023-10-02:
esp32 bumped to 2.2.9
Still 0 sats reported in the meshtastic bluetooth app.

jp-bennett, 2023-10-02:
We're also working an issue with the T-beam 1.2 revision, where the GPS is behaving strangely. Can you check whether your board is a 1.2? There is a label right beside the Lora Antenna connector on the board.

garthvh, 2023-10-03:
Number of sats is a position flag that must be enabled under position settings for it to be sent to the app

closed 2023-10-03 by garthvh

NickDolfin, 2023-10-03:
White silkscreen on PCB near lora antenna:
V22_V1.1
202110222.

Near edge of pcb:
Sticker with CE mark.
FCC ID: 2ASYE-T-BEAM
Model: T-BEAM-V1.1

NickDolfin, 2023-10-03:
@garthvh  Where do you set that flag?
I cannot find it:

[image]

NickDolfin, 2023-10-03:
note to myself: meshtastic --pos-fields SATINVIEW