meshtastic/firmware #2853: [Feature Request]: UA Font for Eink

assada, 2023-10-01:
### Platform

NRF52

### Description

Hi! Made a font for t-echo EInk. Now it is not very convenient to use it due to the replacement of `/firmware/.pio/libdeps/t-echo/ESP8266 and ESP32 OLED driver for SSD1306 displays/src/OLEDDisplayFonts.cpp`

I suggest adding a new font file `EInkDisplayFontsUA.cpp` with `const uint8_t ArialMT_Plain_16_UA[]`. Also to https://github.com/meshtastic/firmware/blob/master/src/graphics/Screen.cpp#L109 add smth like 
```cpp
#ifdef EINK_UA
#define FONT_SMALL ArialMT_Plain_16_UA // Height: 19
#else
#define FONT_SMALL ArialMT_Plain_16 // Height: 19
#endif
``` 

**ArialMT_Plain_16_UA**: https://gist.github.com/assada/8ab477d67653690842c2328faeb19a88

Screenshot: 
[image]

garthvh, 2023-10-01:
Ariel is proprietary, so we probably can’t use it, would need a compatible license.

assada, 2023-10-01:
But it is already in use (ArialMT_Plain_10_UA and ArialMT_Plain_10_RU)

Burnwald, 2023-11-29:
https://github.com/olikraus/u8g2/wiki/fntgrpunifont

closed 2024-03-28 by thebentern