meshtastic/firmware #2862: [Bug]: SX1262 not detected

dizcza, 2023-10-03:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

latest master

### Description

I have a custom board that works fine and RadioLib alone successfully detects an SX1262 chip with the following code taken from examples:

```c
ArduinoHal myhal(SPI);
SX1262 radio = new Module(&myhal, 25, 36, RADIOLIB_NC, RADIOLIB_NC);

...

  int miso = 16;
  int mosi = 2;
  int sck = 15;
  SPI.begin(sck, miso, mosi);
  int state = radio.begin(868);

OUTPUT:
[SX1262] Initializing ... success!
[SX1262] Starting to listen ... success!
```

However, Meshtastic does not recognize the module:

```c
#define USE_SX1262

#define RF95_SCK 15
#define RF95_MISO 16
#define RF95_MOSI 2
#define RF95_NSS 25

#define LORA_RESET RADIOLIB_NC
#define LORA_DIO0 RADIOLIB_NC
#define LORA_DIO1 36
#define LORA_DIO2 39

// common pinouts for SX126X modules
#define SX126X_CS   RF95_NSS    // NSS for SX126X
#define SX126X_DIO1 LORA_DIO1
#define SX126X_BUSY LORA_DIO2
#define SX126X_RESET LORA_RESET
```

The TFT LCD works fine though.

-----

This might be related to https://github.com/meshtastic/firmware/issues/2362 but it seems to be resolved, so I'm reporting a new issue.

### Relevant log output

```Shell
WARN  | ??:??:?? 0 SPI.begin(SCK=15, MISO=16, MOSI=2, NSS=25)
WARN  | ??:??:?? 1 SX126xInterface(cs=25, irq=36, rst=-1, busy=39)
INFO  | ??:??:?? 1 Starting meshradio init...
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 1 (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=253, time 3246 ms
INFO  | ??:??:?? 1 Radio freq=868.375, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 1 Set radio: region=UA_868, name=LongFast, config=0, ch=1, power=14
INFO  | ??:??:?? 1 Radio myRegion->freqStart -> myRegion->freqEnd: 868.000000 -> 868.599976 (0.599976 mhz)
INFO  | ??:??:?? 1 Radio myRegion->numChannels: 2 x 250.000kHz
INFO  | ??:??:?? 1 Radio channel_num: 2
INFO  | ??:??:?? 1 Radio frequency: 868.375000
INFO  | ??:??:?? 1 Slot time: 42 msec
INFO  | ??:??:?? 1 Set radio: final power level=14
INFO  | ??:??:?? 21 SX126x init result -2
WARN  | ??:??:?? 21 Failed to find SX1262 radio
INFO  | ??:??:?? 21 Not using WIFI
ERROR | ??:??:?? 21 NOTE! Recording critical error 3 at src/main.cpp:726
```

dizcza, 2023-10-03:
I must say that RadioLib is no longer able to find the SX1262 after I flash the board with RadioLib example followed by flashing your Meshtastic firmware. Only when I unplug and plug the power back on the RadioLib detects the module again. Maybe you do some greedy enumeration / chip selection that jams the module until a reset.

GUVWAF, 2023-10-03:
One difference that I can see between the configurations is that you set the `SX126X_BUSY` to 39 (`LORA_DIO2`) for Meshtastic, which is the last parameter in RadioLib's `Module`, that you set to `RADIOLIB_NC` there. 
Maybe you could try changing that?

dizcza, 2023-10-03:
After I set `SX126X_BUSY` to `RADIOLIB_NC`, I get the following error:

```
//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (16384/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (120 Bytes)
[   112][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   125][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x42
INFO  | ??:??:?? 0 Device found at address 0x42 was not able to be enumerated
DEBUG | ??:??:?? 0 I2C device found at address 0x4d
INFO  | ??:??:?? 0 Device found at address 0x4d was not able to be enumerated
DEBUG | ??:??:?? 0 I2C device found at address 0x52
INFO  | ??:??:?? 0 Device found at address 0x52 was not able to be enumerated
INFO  | ??:??:?? 0 No I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 Meshtastic hwvendor=42, swver=2.2.10.92b14783
DEBUG | ??:??:?? 0 Setting random seed 2966622186
DEBUG | ??:??:?? 0 Total heap: 186556
DEBUG | ??:??:?? 0 Free heap: 151936
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 95, FreeEntries 535, AllEntries 630, NameSpaces 5
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 50
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x40096f6b
file: "src/platform/esp32/BleOta.cpp" line 16
func: static const esp_partition_t* BleOta::findEspOtaAppPartition()
expression: esp_ota_get_partition_description(part, &app_desc)
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40096f6b
file: "src/platform/esp32/BleOta.cpp" line 30
func: static String BleOta::getOtaAppVersion()
expression: esp_ota_get_partition_description(part, &app_desc)
DEBUG | ??:??:?? 0 No OTA firmware available
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 22
[   306][E][vfs_api.cpp:105] open(): /littlefs/prefs/config.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /prefs/config.proto preferences found
INFO  | ??:??:?? 0 Installing default LocalConfig
INFO  | ??:??:?? 0 Setting default channel and radio preferences!
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 15, using UA_868
[   337][E][vfs_api.cpp:105] open(): /littlefs/prefs/module.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /prefs/module.proto preferences found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
[   390][E][vfs_api.cpp:105] open(): /littlefs/oem/oem.proto does not exist, no permits for creation
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 0 Number of Device Reboots: 50
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 15, using UA_868
DEBUG | ??:??:?? 0 region=15, NODENUM=0x51121994, dbsize=1
INFO  | ??:??:?? 0 Saving /prefs/db.proto
WARN  | ??:??:?? 0 SPI.begin(SCK=15, MISO=16, MOSI=2, NSS=25)
DEBUG | ??:??:?? 0 Read RTC time as 95
DEBUG | ??:??:?? 0 Running without GPS.
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
INFO  | ??:??:?? 0 External Notification Module Disabled
INFO  | ??:??:?? 0 Doing TFT init
INFO  | ??:??:?? 1 Turning on screen
DEBUG | ??:??:?? 1 Module wants a UI Frame
DEBUG | ??:??:?? 1 Module wants a UI Frame
[... 49 lines cut ...]
DEBUG | ??:??:?? 30 [NodeInfoModule] Using AES128 key!
DEBUG | ??:??:?? 30 [NodeInfoModule] ESP32 crypt fr=51121994, num=2b6aa27e, numBytes=50!
DEBUG | ??:??:?? 30 [NodeInfoModule] enqueuing for send (id=0x2b6aa27e fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] txGood=0,rxGood=0,rxBad=0
DEBUG | ??:??:?? 30 [RadioIf] SX126x standby failed with error -707

assert failed: void SX126xInterface<T>::setStandby() [with T = SX1262] SX126xInterface.cpp:225 (err == RADIOLIB_ERR_NONE)

Backtrace: 0x400845d9:0x3ffd97b0 0x40096f8d:0x3ffd97d0 0x4009dcd9:0x3ffd97f0 0x400e6616:0x3ffd9920 0x400e658c:0x3ffd9940 0x400ea913:0x3ffd9960 0x400ea97c:0x3ffd9980 0x4022656a:0x3ffd99a0 0x400d95a5:0x3ffd99c0 0x400d9672:0x3ffd99e0 0x40104bf5:0x3ffd9a00 0x400e465a:0x3ffd9a30 0x4013a7d9:0x3ffd9a50

  #0  0x400845d9:0x3ffd97b0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x40096f8d:0x3ffd97d0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4009dcd9:0x3ffd97f0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x400e6616:0x3ffd9920 in SX126xInterface<SX1262>::setStandby() at src/mesh/SX126xInterface.cpp:225
  #4  0x400e658c:0x3ffd9940 in SX126xInterface<SX1262>::isChannelActive() at src/mesh/SX126xInterface.cpp:281
  #5  0x400ea913:0x3ffd9960 in RadioLibInterface::onNotify(unsigned int) at src/mesh/RadioLibInterface.cpp:219
      (inlined by) RadioLibInterface::onNotify(unsigned int) at src/mesh/RadioLibInterface.cpp:194
[... cut ...]
  #8  0x400d95a5:0x3ffd99c0 in concurrency::NotifiedWorkerThread::runOnce() at src/concurrency/NotifiedWorkerThread.cpp:89
  #9  0x400d9672:0x3ffd99e0 in concurrency::OSThread::run() at src/concurrency/OSThread.cpp:85
  #10 0x40104bf5:0x3ffd9a00 in ThreadController::runOrDelay() at .pio/libdeps/ut2uh-k46v2/Thread/ThreadController.cpp:59
  #11 0x400e465a:0x3ffd9a30 in loop() at src/main.cpp:798
  #12 0x4013a7d9:0x3ffd9a50 in loopTask(void*) at /home/dizcza/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

dizcza, 2023-10-03:
Hm, after I set the last gpio RadioLib.Module arg to 39 and replug the device, it's also no longer recognized in RadioLib.

And I do see the first log of yours as a success, suggesting that RadioLib's success could also be fake. I need a second device to verify that.

GUVWAF, 2023-10-03:
It's indeed recognized correctly now, but it fails to set it to standby (which might not be done with the simple sketch from RadioLib).
In [the `standby()` function](https://github.com/jgromes/RadioLib/blob/master/src/modules/SX126x/SX126x.cpp#L479), it pulls the CS pin low, which is otherwise not often used. Can you check whether `SX126X_CS` set to 25 is actually correct?

dizcza, 2023-10-03:
> Can you check whether `SX126X_CS` set to 25 is actually correct?

Yes, it's set correctly to 25.

mverch67, 2023-10-04:
In your first LOG you had the correct output:

`SX126xInterface(cs=25, irq=36, rst=-1, busy=39)`

but in the second LOG the busy is wrong which causes the failure in the standby.

`SX126xInterface(cs=25, irq=36, rst=-1, busy=-1)`

dizcza, 2023-10-04:
> If it fails with the busy signal set correctly it maybe that the wiring is not correct for this.

Yes, it fails with the busy signal set correctly.

All right, I'll have a look at the wiring. Thanks for the hint.

dizcza, 2023-10-08:
I added
```c
  int err_code = radio.standby();
  Serial.printf("standby %d\n", err_code);
```
to a RadioLib sketch I showed in the header with the busy pin set to `RADIOLIB_NC` and it works fine:
```
standby 0
```
So the issue is probably on the Meshtastic side.

dizcza, 2023-10-08:
Lora1262 NiceRF https://www.nicerf.com/item/868mhz-sx1262-lora-module-lora1262
ESP32
Not sure if I can upload the schematics other than the pins I've specified. Here is the variant file:

```c

#define I2C_SDA 21
#define I2C_SCL 22

#define BUTTON_PIN 13
#define BUTTON_NEED_PULLUP // if set we need to turn on the internal CPU pullup during sleep

#define USE_SX1262

#define RF95_SCK 15
#define RF95_MISO 16
#define RF95_MOSI 2
#define RF95_NSS 25

#define LORA_RESET RADIOLIB_NC
#define LORA_DIO0 RADIOLIB_NC
#define LORA_DIO1 36
#define LORA_DIO2 39

// common pinouts for SX126X modules
#define SX126X_CS   RF95_NSS    // NSS for SX126X
#define SX126X_DIO1 LORA_DIO1
#define SX126X_BUSY LORA_DIO2
#define SX126X_RESET LORA_RESET

// Set lora.tx_power to 13 for Hydra or other E22 900M30S target due to PA
#define SX126X_MAX_POWER 22

// This board has different GPS pins than all other boards
#define HAS_GPS 0
#define GPS_RX_PIN 0
#define GPS_TX_PIN 0

// Define if screen should be mirrored left to right
#define SCREEN_ROTATE

// LCD screens are slow, so slowdown the wipe so it looks better
#define SCREEN_TRANSITION_FRAMERATE 1 // fps

#define TFT_HEIGHT 240
#define TFT_WIDTH 320
#define TFT_OFFSET_X 0
#define TFT_OFFSET_Y 0
#define TFT_BUSY -1

#define ILI9341_SPI_HOST VSPI_HOST // VSPI_HOST or HSPI_HOST
```

dizcza, 2023-10-08:
Here is the relevant part of the schematics:
[image]

dizcza, 2023-10-20:
Yeah sorry for the late response.

> The module you linked to is an SX1262 but the module in your schematic is not the same, there are pins in your schematic which don't exist

This is because the pinout socket we're using supports multiple types of LoRa chips: 126x, 127x, ...

>  #define SX126X_CS 36
  #define SX126X_DIO1 36

You probably meant different pins here.

Anyhow, the problem could lie in the capacitor value that connects the RESET to VCC and GND. We need to check this one first.

closed 2023-10-20 by dizcza