meshtastic/firmware #2868: [Bug]: gps coordinates does not change

NickDolfin, 2023-10-05:
### Category

Other

### Hardware

T-Beam

### Firmware Version

2.2.10.7cebd79

### Description

I have a Tbeam T22_V1.1 / 20210222.

After aquiring a gps coordinate, it will nit change anymore.
gps update interval: 30s
send out over mesh: 180s
fix attempt duration: 900s

Any idea?

jp-bennett, 2023-10-05:
Can you capture a log output using Meshtastic --noproto? Need to see what happens when it tries to get an updated GPS lock.

NickDolfin, 2023-10-05:
C:\Users\nick\Desktop\Research\electronics\Lora\meshtastic> Meshtastic --noproto
WARNING file:mesh_interface.py _sendToRadio line:680 Not sending packet because protocol use is disabled by noProto
Connected to radio
WARNING file:mesh_interface.py _sendPacket line:530 Not sending packet because protocol use is disabled by noProto
INFO  | 21:24:55 2513 [DeviceTelemetryModule] (Sending): air_util_tx=1.205278, channel_utilization=0.000000, battery_level=100, voltage=4.153000
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] updateTelemetry LOCAL
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Node status update: 2 online, 2 total
INFO  | 21:24:55 2513 [DeviceTelemetryModule] Sending packet to mesh
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Update DB node 0xf71ea694, rx_time=1696541095, channel=0
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] handleReceived(LOCAL) (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1696541095 priority=1)
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] No modules interested in portnum=67, src=LOCAL
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] localSend to channel 0
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Add packet record (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1696541095 priority=1)
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Expanding short PSK #1
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Using AES128 key!
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] ESP32 crypt fr=f71ea694, num=25e0c71f, numBytes=23!
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] enqueuing for send (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1696541095 priority=1)
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] txGood=59,rxGood=59,rxBad=0
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] NOTE: tophone queue status queue is full, discarding oldest
WARN  | 21:24:55 2513 [DeviceTelemetryModule] ToPhone queue is full, discarding oldest
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Using channel 0 (hash 0x8)
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Expanding short PSK #1
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] Using AES128 key!
DEBUG | 21:24:55 2513 [DeviceTelemetryModule] ESP32 crypt fr=f71ea694, num=25e0c71f, numBytes=23!
DEBUG | 21:24:56 2513 [DeviceTelemetryModule] decoded message (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1696541095 priority=1)
DEBUG | 21:24:56 2513 [RadioIf] Starting low level send (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1696541095 priority=1)
DEBUG | 21:24:56 2513 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=39, time 698 ms
DEBUG | 21:24:56 2513 [RadioIf] AirTime - Packet transmitted : 698ms
DEBUG | 21:24:56 2514 [RadioIf] Completed sending (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1696541095 priority=1)
DEBUG | 21:24:59 2517 [RadioIf] (bw=250, sf=11, cr=4/8) packet symLen=8 ms, payloadSize=39, time 698 ms
DEBUG | 21:24:59 2517 [RadioIf] Lora RX (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=9 rxRSSI=-51)
DEBUG | 21:24:59 2517 [RadioIf] AirTime - Packet received : 698ms
DEBUG | 21:24:59 2517 [Router] Rx someone rebroadcasting for us (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=9 rxRSSI=-51)
DEBUG | 21:24:59 2517 [Router] didn't find pending packet
DEBUG | 21:24:59 2517 [Router] Found existing packet record for fr=0xf71ea694,to=0xffffffff,id=0x25e0c71f
DEBUG | 21:24:59 2517 [Router] Found existing packet record for fr=0xf71ea694,to=0xffffffff,id=0x25e0c71f
DEBUG | 21:24:59 2517 [Router] Add packet record (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=9 rxRSSI=-51)
DEBUG | 21:24:59 2517 [Router] Ignoring incoming msg, because we've already seen it (id=0x25e0c71f fr=0x94 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=9 rxRSSI=-51)
DEBUG | 21:24:59 2517 [Router] cancelSending id=0x25e0c71f, removed=0
DEBUG | 21:24:59 2517 [Router] Incoming message was filtered 0xf71ea694
DEBUG | 21:25:03 2520 [GPS] WANT GPS=1
INFO  | 21:25:03 2520 [GPS] Setting GPS power=1
WARN  | 21:25:03 2520 [GPS] SOME data is TOO OLD: LOC 30052, TIME 30052, DATE 30052
WARN  | 21:25:03 2520 [GPS] SOME data is TOO OLD: LOC 30259, TIME 30259, DATE 30259
WARN  | 21:25:03 2521 [GPS] SOME data is TOO OLD: LOC 30478, TIME 30478, DATE 30478
WARN  | 21:25:03 2521 [GPS] SOME data is TOO OLD: LOC 30696, TIME 30696, DATE 30696
DEBUG | 21:25:03 2521 [GPS] WANT GPS=0
DEBUG | 21:25:03 2521 [GPS] GPS Lock took 0, average 1
INFO  | 21:25:03 2521 [GPS] Setting GPS power=0
DEBUG | 21:25:03 2521 [GPS] publishing pos@651f29b0:2, hasVal=1, Sats=5, GPSlock=1
DEBUG | 21:25:03 2521 [GPS] onGPSChanged() pos@651f29b0, time=1696541103, lat=520283506, lon=56484741, alt=25
INFO  | 21:25:03 2521 [GPS] updatePosition LOCAL pos@651f29b0, time=1696541103, latI=520283506, lonI=56484741, alt=25
DEBUG | 21:25:03 2521 [GPS] Node status update: 2 online, 2 total
DEBUG | 21:25:06 2523 [Power] Battery: usbPower=1, isCharging=0, batMv=4155, batPct=100
INFO  | 21:25:16 2533 [PowerFSM] Turning off screen
DEBUG | 21:25:26 2543 [Power] Battery: usbPower=1, isCharging=0, batMv=4155, batPct=100
DEBUG | 21:25:33 2551 [GPS] WANT GPS=1
INFO  | 21:25:33 2551 [GPS] Setting GPS power=1
WARN  | 21:25:33 2551 [GPS] SOME data is TOO OLD: LOC 30052, TIME 30052, DATE 30052
WARN  | 21:25:34 2551 [GPS] SOME data is TOO OLD: LOC 30259, TIME 30259, DATE 30259
[... cut ...]
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\nick\AppData\Roaming\Python\Python311\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Users\nick\AppData\Roaming\Python\Python311\site-packages\meshtastic\__main__.py", line 1351, in main
    common()
  File "C:\Users\nick\AppData\Roaming\Python\Python311\site-packages\meshtastic\__main__.py", line 975, in common
    time.sleep(1000)
KeyboardInterrupt
^C
C:\Users\nick\Desktop\Research\electronics\Lora\meshtastic>

NickDolfin, 2023-10-05:
above is from 2.2.9
I downgraded, but it seems to have a similar result.

NickDolfin, 2023-10-05:
and now with 2.2.10:

Connected to radio
DEBUG | 21:48:30 41 [Power] Battery: usbPower=1, isCharging=0, batMv=4155, batPct=100

DEBUG | 21:48:33 45 [GPS] WANT GPS=1

INFO  | 21:48:33 45 [GPS] Setting GPS power=1

WARN  | 21:48:33 45 [GPS] SOME data is TOO OLD: LOC 30426, TIME 30426, DATE 30430

WARN  | 21:48:34 45 [GPS] SOME data is TOO OLD: LOC 30634, TIME 30634, DATE 30638

WARN  | 21:48:34 45 [GPS] SOME data is TOO OLD: LOC 30853, TIME 30853, DATE 30857

WARN  | 21:48:34 46 [GPS] SOME data is TOO OLD: LOC 31072, TIME 31072, DATE 31076

WARN  | 21:48:34 46 [GPS] BOGUS course.value() REJECTED: -1499236378

DEBUG | 21:48:34 46 [GPS] WANT GPS=0

DEBUG | 21:48:34 46 [GPS] GPS Lock took 0, average 0

INFO  | 21:48:34 46 [GPS] Setting GPS power=0

DEBUG | 21:48:34 46 [GPS] publishing pos@651f2f32:2, hasVal=1, Sats=4, GPSlock=1

DEBUG | 21:48:34 46 [GPS] onGPSChanged() pos@651f2f32, time=1696542514, lat=52x, lon=56x, alt=30

INFO  | 21:48:34 46 [GPS] updatePosition LOCAL pos@651f2f32, time=1696542514, latI=52x, lonI=56x, alt=30

DEBUG | 21:48:34 46 [GPS] Node status update: 2 online, 2 total

INFO  | 21:48:35 46 [DeviceTelemetryModule] (Sending): air_util_tx=0.047639, channel_utilization=8.343333, battery_level=100, voltage=4.155000

DEBUG | 21:48:35 46 [DeviceTelemetryModule] updateTelemetry LOCAL

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Node status update: 2 online, 2 total

INFO  | 21:48:35 46 [DeviceTelemetryModule] Sending packet to mesh

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Update DB node 0xf71ea694, rx_time=1696542515, channel=0

DEBUG | 21:48:35 46 [DeviceTelemetryModule] handleReceived(LOCAL) (id=0x450da00a fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1696542515 priority=1)

DEBUG | 21:48:35 46 [DeviceTelemetryModule] No modules interested in portnum=67, src=LOCAL

DEBUG | 21:48:35 46 [DeviceTelemetryModule] localSend to channel 0

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Add packet record (id=0x450da00a fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1696542515 priority=1)

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Expanding short PSK #1

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Using AES128 key!

DEBUG | 21:48:35 46 [DeviceTelemetryModule] ESP32 crypt fr=f71ea694, num=450da00a, numBytes=28!

DEBUG | 21:48:35 46 [DeviceTelemetryModule] enqueuing for send (id=0x450da00a fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1696542515 priority=1)

DEBUG | 21:48:35 46 [DeviceTelemetryModule] txGood=2,rxGood=4,rxBad=0

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Using channel 0 (hash 0x8)

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Expanding short PSK #1

DEBUG | 21:48:35 46 [DeviceTelemetryModule] Using AES128 key!

DEBUG | 21:48:35 46 [DeviceTelemetryModule] ESP32 crypt fr=f71ea694, num=450da00a, numBytes=28!

DEBUG | 21:48:35 46 [DeviceTelemetryModule] decoded message (id=0x450da00a fr=0x94 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 rxtime=1696542515 priority=1)
[... 824 lines cut ...]
WARN  | 21:54:14 385 [GPS] SOME data is TOO OLD: LOC 30057, TIME 30057, DATE 30057

WARN  | 21:54:14 385 [GPS] SOME data is TOO OLD: LOC 30265, TIME 30265, DATE 30265

WARN  | 21:54:14 385 [GPS] SOME data is TOO OLD: LOC 30484, TIME 30484, DATE 30484

WARN  | 21:54:14 386 [GPS] SOME data is TOO OLD: LOC 30703, TIME 30703, DATE 30703

WARN  | 21:54:14 386 [GPS] BOGUS course.value() REJECTED: -1499236378

DEBUG | 21:54:14 386 [GPS] WANT GPS=0

DEBUG | 21:54:14 386 [GPS] GPS Lock took 0, average 0

INFO  | 21:54:14 386 [GPS] Setting GPS power=0

DEBUG | 21:54:14 386 [GPS] publishing pos@651f3086:2, hasVal=1, Sats=4, GPSlock=1

DEBUG | 21:54:14 386 [GPS] onGPSChanged() pos@651f3086, time=1696542854, lat=52x, lon=56x, alt=30

INFO  | 21:54:14 386 [GPS] updatePosition LOCAL pos@651f3086, time=1696542854, latI=52x, lonI=56x, alt=30

DEBUG | 21:54:14 386 [GPS] Node status update: 2 online, 2 total

DEBUG | 21:54:30 401 [Power] Battery: usbPower=1, isCharging=0, batMv=4155, batPct=100

NickDolfin, 2023-10-05:
I hope you can find something.

Many thanks.

NickDolfin, 2023-10-05:
[image]

After a while I see the above picture on my ipad

thebentern, 2023-10-10:
Should be fixed in 2.2.11. Please test and reopen if not resolved

closed 2023-10-10 by thebentern

NickDolfin, 2023-11-05:
Hi,

I just reopened this (I think).

After flashing version 2.2.11 I still get wrong positions:

About half of the time I get an altitude of 42.950 km high.